opentx/opentx #8654: Cannot flash fw under linux OS with companion23

mozgy, 2021-08-05:
Hello,

opentx 2.3.14, Fedora 34 Qt source rebuilt companion23
radiomaster tx16s

per documentation if you connect tx in bootloader mode it'll be in USB storage -
`Bus 001 Device 018: ID 0483:5720 STMicroelectronics Mass Storage Device`
so dfu-util won't find any device and will not work
```
# /usr/bin/dfu-util -a 0 --dfuse-address 0x08000000:2097152 --device 0483:df11 -U/tmp/16850-flash-check.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available
```

editing radio models works fine

jtaylor2, 2021-08-06:
Transmitter is in DFU mode when it is connected power off.  Connect it powered off to flash with DFU and in boot manager to read/write models and settings

3djc, 2021-08-06:
Indeed radio needs to be turned off for dfu flash

closed 2021-08-06 by 3djc

mozgy, 2021-08-06:
Yeah, my bad, was following errorneuss web page.