opentx/opentx #8655: Make text in syncing between radio and computer as it was before in 2.2.4

nostromo123, 2021-08-06:
For a normal user, who only syncs between radio and local from time to time, the 'old' text was in my eyes much cleared. 

The question, what is source and what is destination is not clear, radio and local is much easier to get.

So please think of returning to the 'old' text.

br KH

[image]

therealzanfar, 2021-08-18:
I agree that the new text is decidedly unclear--it appears as if a generic synchronization dialog has simply had the local/radio folder labels slapped on. I'm not sure the "old" text is the clearest--I'm not completely sure what "first" means here--but the use of "Local" and "Radio" vs "Source" and "Destination" at least match the corresponding folder selection labels.

nostromo123, 2021-08-18:
The problem, in my eyes, is here, that a user programming mainly with the radio sees the radio as 'source' and the computer running companion as 'destination', whereas a user doing the programming in companion sees the situation vice versa. Absolutely not clear, as it is now.

br KH

elecpower, 2021-08-18:
Duplicate of #8319

closed 2021-08-18 by elecpower