opentx/opentx #8657: Download a firmware doesn't work

0-Sachatte, 2021-08-07:
OpenTX V2.3.14 on Windows 10 21H1

Hello, I got a bug on Windows I can't download a firmware. 

When I got to download then "Download firmware" it open a file explorer to select where to put it.
Then I select "Save" of the file explorer and I got a error because the file do not exist : 
[image]
And when I create a file with the name that it want, it will not change the data in it (I got a 0 bytes file)

I've tried on Pop OS 20.04 (which is a Ubuntu) and all works fine.

kilrah, 2021-08-07:
Sounds like you're trying to save to a place with no write permissions.

0-Sachatte, 2021-08-08:
Ok, you were right. The program didn't got the write permission on my folder.
Thanks.

closed 2021-08-08 by 0-Sachatte