opentx/opentx #8658: OpenTx Update 2.3.14 Multi module Issue Horus X10 Express

youngstar5, 2021-08-08:
Just updated openTX to 2.3.14 and module Isrm to V2.1.6 along with receivers to latest firmware R9SX & RX4R as I am using two receivers for dual frequency redindancy. All has been working fine until these new updates. 
It would appear the failsafe no longer functions. I try each receiver independantly by turning off connection and the failsafe per receiver works fine. As soon as I use the two receivers and modules no failsafe, any ideas ? 

I havent tried the process of elimination by backward replacing updates for isrm to TX 2.1.0
Just to first pose my issues and see if it can be replicated.

Specs.: Modules: R9M 2019 Access,  ISRM V2.1.6 
Rx: R9 SX V1.3.3 : RX4R (Access) V2.1.7 

Failsafe mode selected in all cases: Custom

brainbubblersbest, 2021-08-10:
When Do you expect the failsave with dual receivers in redundancy?
With loss of connection to a Single receiver, or with loss of both connetions?

youngstar5, 2021-08-10:
With Transmission failure obviously This is the same argument with two receivers on the same frequency. Failsafe setting should still apply as we are still relying on an Rf signal.
It could be argued that the second receiver is your failsfae but if the Transmitter total failure for whatever reason then a failsafe should kick in. Given in the menu of the TX both modules allow for custom settings. I am sure the clever software folk could write some code for either strongest signal received last or both modules with same custom settings in failsafe to act as failsafe?
Perhaps not enough demand for this but it would make Frsky a leading brand in this area as far as I am aware and I aint no expert. The system is fantastic as it is so adding this simply surpasses the competition, progress!
If not possible then just a thought...

brainbubblersbest, 2021-08-10:
Sorry, it's still Hard to understand your Problem.
If you are using r9 and 2,4 in redundancy. Then it's normal failsave kick not in after only one hf connection Fails.

Thats what "Redundancy" is.

Only in the case that all two Systems loosing connection, Failsave need to kick in. 

How Do you have r9sx connected to rx4r?

3djc, 2021-08-11:
I do not have any access equipement to test that, but at first look, there is nothing in the code preventing custom failsafe to be send when two modules are in place

kilrah, 2021-08-11:
What version were you running on the receivers/module before? Pretty sure there have been changes/breaks in failsafe behavior when going from 1.x to 2.x. Might be worth reverting either OpenTX or module/receiver to validate which has the issue.

youngstar5, 2021-08-11:
Hi, many thanks for the comments.
The strange thing is the recivers are simply acting as failsafe set to Hold when TX turned off and the failsafes have been set to Custom with settings on each module.
I tried to reconfigure backwards to no avail but I definately had one of my models working ok with the two receivers and respective modules.
The firwares used then were:
Receivers -  R9sx V1.3.2 (R9SX_LBT.frsk / RX4R (access) V2.1.0 
Modules -  R9M2019 Access V1.3.0 (RM2019_LBT.frk) No updates on this module? / ISRM V2.1.0 
My current setup is as my first post. If the code looks ok it may suggest given no update on R9M2019 module! I dont know.

I do know that turning each Tx module off in turn results in correct Failsafe action.

brainbubblersbest, 2021-08-11:
Do you mean turning each tx module of in turn results in correct failsave Action. - > This would be correct then

Or "should" return in correct failsave Action. - >This would describe that you are expect it this way, but it's not happening. 

It's still unclear what you have connected to your receivers. 
Are you using a Flight Controller? 

Or have you Servos connected to the RX4R? 

You need to set Failsave to "No Pulses" with an FC. 
The FC then detects an dead Receiver or lost connection to it. 
Also the second Receiver detect a lost connection to the first in this way. 
If you Set Failsave with custom Channel output, then the other receiver still receive valid Channels.

brainbubblersbest, 2021-08-11:
Btw. You can use a r9mm or r9slim+ for the redundancy setup receiver. Less weight and less money.

youngstar5, 2021-08-30:
No Flight Controller. SBus in /Out
3djc your right I think both receivers are sending a failsafe and it is cancelling out the (identical) Custom settings in both receivers such that a hold is maintained on all servos.

brainbubblersbest, 2021-08-31:
Sorry, You literally dont understand what i wrote above
The receiver you connect to your servos need to be set with your custom failsave settings.

The other need to be Set to no pulses to maintain your other one detects a failsave.

Thats just logical. How should your Main  receiver detect a lost connection from your Backup receiver, if the Backup receiver send still valid channels (of course custom failsave channel output) over sbus?

Why dont you just try it?
Or should i make a Video with a proof of work just to Show you it works as expected?

Please read my previous answer again. Think through it. Understand it. Test it.

robustini, 2021-09-07:
There is another related problem that I encountered yesterday, always related to the "isrm pulse" issue, where a fix was made but probably did not solve the problem 100%.
In a big glider I install two G-RX8s in ACCESS with the latest firmware installed (2.1.7), updated ISRM module too.
Everything works perfectly, but if I enable the MPM to use it as a trainer, every now and then it releases the student.
MPM loses the synchronism from the TX, the LED starts flashing and then the link becomes active again randomly.
I therefore believe that the latest fix to the isrm pulse needs an additional fix.

robustini, 2021-09-07:
There is another related problem that I encountered yesterday, always related to the "isrm pulse" issue, where a fix was made but probably did not solve the problem 100%.
In a big glider I install two G-RX8s in ACCESS with the latest firmware installed (2.1.7), updated ISRM module too.
Everything works perfectly, but if I enable the MPM to use it as a trainer shortly after the link with the student is dropped.
MPM loses the synchronism from the TX, the LED starts flashing and then the link becomes active again randomly.
Switching the TX off and on again, the link is always re-established, but it is not stable. 
I therefore believe that the latest fix to the isrm pulse needs an additional fix.

AlphaCharlyX, 2021-09-22:
Maybe you have the same issue I have:

[https://github.com/opentx/opentx/issues/8708](https://github.com/opentx/opentx/issues/8708)

youngstar5, 2021-10-04:
Hi Brainbubble, it doesnt apply to this issue, I followed your action as you described - results in 'No Changes' for my issues.
Unless you have the same hardware set up as I above then your interpretation does not apply. Thank you anyway.

Hi AlphaCharlyX, thank you not sure but there is certainly an issue with either receivers or ISRM module update. Either way I am reverting to previouse software as I certainly cant go anywhere forwards with no failsafe, now a legal requirement in UK though hold maybe legal !

brainbubblersbest, 2021-10-06:
Which setup are you using.
FC or Servos?
A or B?
[image]
[image]

youngstar5, 2021-12-06:
Hi,
I now know what is happening with the failsafe. 
Sorry Brainbubble but not as you describe for my situation (B in your Diag.) as previously mentioned.
You certainly do not disable the failsafe settings for one of the receivers. When you first apply one receiver failsafe settings the Horus automatically appliess the same setting to the second receiver.  logic being if either frequency receiver dies then the other receiver picks up seamlessly and should this receiver also lose a signal then its failsafe settings apply. This does work, so on no account disable one receivers failsafe settings with two receivers on different frequencies using the Horus Express & one receiver driving servos and connection via sbus as brainbubble has made the effort to show in diag. B ! Else if your end up flying on the second backup (slave RX) then should their be an RF issue you will have no failsafe.
Great system pitty Frsky could not have pointed this out more clearly, such a great selling point.

closed 2021-12-06 by kilrah