opentx/opentx #8661: Default channel order for new models being ignored

jdosher, 2021-08-11:
When changing the default channel order in the radio profile, the new channel order is being ignored when creating new models (default or via wizard). 

To reproduce:

1. create a new otx file
2. create a new model, note that the channel order follows the default set in the radio profile
3. open the radio profile and change the channel order
4. create a new model, note that the channel order does NOT follow the newly specified order, but instead follows the original order

Expected behavior:
Newly created models should follow the channel order set in the radio profile.

Observed behavior:
Newly created models do not follow the channel order set in the radio profile, but instead follow the channel order that existed when the open otx file was created.

Comments:
It appears that if you change the channel order in a radio profile that setting is ignored, and instead the channel order that existed when the otx file was created overrides the setting in the radio profile. Note that this is about newly created models - it's understood that existing models do not update their channel order when the default order is changed. 

[image]

timtuxworth, 2021-08-11:
Yes I have seen this, I was confused by what I saw, it didn't seem right, now I know why.

kilrah, 2021-08-11:
Normal behavior, was explained in your forum topic.

https://openrcforums.com/forum/viewtopic.php?f=94&t=13520&p=154496#p154496

closed 2021-08-11 by kilrah

jdosher, 2021-08-11:
Really? That's certainly not what's described in the documentation. The
behaviour I've always read is that changing the default channel order in
the radio profile will then change the channel order in newly created
models. No where in the documentation have I read that you have to create
an entirely new otx file in order for the default channel order to update.

kilrah, 2021-08-11:
Both the default stick mode and default channel order in the profile only apply to new documents, it's always been like that. In what documentation have you seen otherwise?

jdosher, 2021-08-11:
I guess the confusion is that for all intents and purposes it appears to
the user that they're successfully changing the default channel order in
the document they have open because there's no indication otherwise. The
drop-down isn't grayed out, and there's nothing informing the user the
effect only applies to a new, blank otx. So I can see how the current
behaviour is not a bug per-se, but maybe a UX issue?

kilrah, 2021-08-12:
Why would it be grayed-out? You can create a new document at any time and it'll have that order.

The radio has to take the default channel order into account, it doesn't know what companion is set to, so it's kind of obvious the setting used to make a new model is the one the radio has access to i.e. the one that's in the document.

jdosher, 2021-08-12:
Well, all I can say is this: all over the various forums and YouTube videos
you see the same notion repeated: changing the default channel order in the
radio profile only changes newly created models, not your existing models.
If someone creates a new, blank otx file, obviously there are no existing
models. So clearly users are confused by the effect of changing the default
channel order - you even saw someone express that they were confused by
this in the forum posting I made.

Yes, you can say those users are wrong.

But if a lot of users are confused by a feature, it might be a confusing
implementation of that feature. My suggestion was to gray out the dropdown
if a populated otx file is open - because changing it will have NO effect
on newly created models in that file, but many, many users are under the
misguided belief that it does. Or my other suggestion was to raise a dialog
if the default channel order is changed while a populated (non-blank) otx
file is already open, alerting the user that the change will only take
effect when they create a new otx file.

I'm not an OTX dev (I have been using Qt since the late 90's however) - I'm
a user, but I'm telling you that there's a feature that's implemented in a
way that's causing confusion to a lot of users and it might be an indicator
that a UX change is needed.

kilrah, 2021-08-12:
> all over the various forums and YouTube videos
you see the same notion repeated: changing the default channel order in the
radio profile only changes newly created models, not your existing models.

Can you point to one?
I think you're continuously confusing the companion radio profile with the radio general settings since the videos I know of refer to the latter. Or maybe some of them do use the wrong name, but when you give 2 names to different things you can't exactly force people not to use the wrong one...

jdosher, 2021-08-12:
I'm talking about the default channel order dropdown box in the radio
profile settings in Companion: Settings > Settings > Radio Profile tab on
the 'Edit Settings' window that pops up.

At this point, I'm not sure that sending you examples of people incorrectly
describing the behaviour is going to do any good to convince you. I
presented some options to help ease users' confusion. I don't see any point
in us continuing to go back and forth on this if you disagree.

Thanks for your work on OpenTX and Companion; they're a great resource.

-Jesse

timtuxworth, 2021-08-13:
How about this? When the user changes the default order in the Radio Profile - prompt them "do you want this to apply to your open radios", also when opening an otx file or loading from the radio - check if the channel order is different from the radio profile and if it is ask the user "Your default channel order in the radio profile is different to the one in this radio, do you want to update it?"