opentx/opentx #8663: dsm external module could not bind after  openTx 2.3.11

Nesan65, 2021-08-15:
dsm external module could not bind after  openTx 2.3.11
"the dsm module made out  of spectrum tx".  works fine in opentx 2.3.11.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

3djc, 2021-08-15:
don't have x12 to test, but logic analyser look at tx16s between 2.3.11 and 2.3.14 shows no differences

Nesan65, 2021-08-15:
As I said the external module is the one build from original DX6i TX it has no bide button so I can bide only in OpenTx. was working flawless until 2.3.11, any after it dose not work. I rolled back to 2.3.11 it works again I even try in Horus X10S radio; same result. 
is there any binding sequence in new opentx like holding sticks in cretin position. for initiate bind sequence?  "in DSM mode"

Here I would like to say the regular multi protocol module works fine though.

Linseder, 2022-09-11:
Radio Horus X12S ACCESS ISRM upgraded with External access mod.

The problem with the DSM external module is still not solved.
When binding with Opentx in DSMX mode, there is a query for "CH1-8 Telem On" or "CH1-8 Telem OFF", as far as I know, this module can only do channels 1 to 6. 
Tested with Opentx 2.3.15.
Multi protocol module works fine.

Nesan65, 2022-09-14:
All work in opentx 2.3.15. I tested orginal spektrum module and multiprotocol  module as well. If orginal spektrum module not bind. Try first to call binding the Tx then with binding plug on the Rx. It will bind.