opentx/opentx #8667: ADC filter

MrShadey, 2021-08-17:
It is advised by the betaflight Dev's to turn off the ADC filter in hardware,as it is dealt with in software. However it is advisable to leave it turned on for models that have servos to stop them from constantly twitching and potentially wearing out the potentiometer and gearing prematurely. 
Can the ADC filter setting be made model specific. To avoid the need to turn it on or off when switching between models with flight controllers and planes or heli's that use servos and no software ADC filter.

mituritsyn, 2021-12-17:
#8396

Lexiyntax, 2022-07-10:
Duplicate of #8396

Lexiyntax, 2022-07-10:
Note as per #8396 this feature seems to have a solution in the process of being merged in EdgeTX already, https://github.com/EdgeTX/edgetx/pull/1363