opentx/opentx #8670: Master Jack not recognized under 2.3.14

ChrisOhara57, 2021-08-19:
After installing 2.3.14 on x9lite S the slave is not recognized via master/jack for Trainer functionality.
Did work with same constellation /s ame model on 2.3.7.
Also calibration not possible.

mStrangers, 2021-08-31:
Same issue with a qx7,qx7S,X9 lite and x9 liteS master work on 2.3.11 but not on newer version.
But no issues on X9D+ 2019 and SE , and XliteS and Xlite Pro the master/jack work on 2.3.14

mStrangers, 2021-08-31:
mine are qx7 access and qx7s access tested on 3-4 model of each and i guaranty, slave work but not master

ChrisOhara57, 2021-08-31:
Just a hint. I do not know whether it matters.
In your site about OpenTx 2.3.14   ( https://www.open-tx.org/2021/07/23/opentx-2.3.14 )
under "new since 2.3.13" I found this:

[image]

May be the same problem was fixed for x9D but it was not known that there was also a problem with other radios??

mariusvds, 2021-09-27:
I do have a TX16S and a QX7S, slave is not working in 2.3.14 on both radios (it did in 2.3.10), I checked it by hooking them up to each other, both of them does not recognize the signal of the other, it seems to be a slave issue, because I do have a AeroSIM RC dongle, which worked with OTX 2.3.10, but not with 2.3.14, so it has to be in the slave side

mariusvds, 2021-09-28:
> I just tested TX16S and X7 both with 2.3.14, it works absolutely fine both ways:

Yeah, I was confused by two problems arising at the same time, check my edit

ChrisOhara57, 2021-09-28:
Hi
I have thoroughly tested and you can believe me:
  my x9lite S works  perfect with 2.3.7 and 2.3.11 master/jack  and slave/jack
  but with 2.3.14 master/jack is not working   ---- but slave jack is working fine with simulator dongle

Is it possible that the varying behavior under 2.3.14  is caused by different setups of the radios?

I do use:   lua; english menu; german speech, EU-LBT, Access
                bluetooth is set to telemetry and switched on

3djc, 2021-11-10:
Most likely the wrong cable used, or wired the wrong way. Previously pin detection was not used

ChrisOhara57, 2021-11-10:
> Most likely the wrong cable used, or wired the wrong way. Previously pin detection was not used

Please give me a hint.
How can I use the wrong cable or wire the wrong way?
I have tested three audio cables (one mono two stereo) both do work on 2.3.7 and on 2.3.11

justtrashmaker, 2021-11-10:
If it works with the old version (2.3.9), but no longer with 2.3.14, the cable cannot be wrong. 

I took a closer look at the hardware. If I bridge the switching contact on the socket (DSC), it also works with 2.3.14. The contact detection seems to be active.

ChrisOhara57, 2021-11-11:
Hello
My knowledge of electronics is rather limited.
From the comment of 3djc "Previously pin detection was not used" I guess that there were changes in 2.3.14 firmware.
But I have no idea what that means to the cable or the switching contact of the socket.
How does pin detection work?

justtrashmaker, 2021-11-11:
It doesn't matter how the detection works. In the old version, the trainer worked perfectly. So something must have changed in the software. With the new software, I also no longer have the option of operating the trainer via SBUS.
That was also different with the old version (2.3.9). I'm not a programmer, but I think if those in charge look at the old version and compare it to the new one, they should be able to see the difference and fix it.

shaun190, 2021-11-13:
This is a problem that’s bugging me too.

I have two brand new QX 7 Access/ ACCST txs working on 2.3.14 that will not work as master transmitters, only ever as a slave tx.
If I use an older QX7 running 2.2.4 I have no problems at all with any cable I pick or using this tx as a master with either of the other txs as slaves.
I have also tried using an iRangeX MPM with a Radiomaster WT01 wireless trainer and that won’t make one of these newer QX 7s work as a master either.  
I have also tried using a Radiomaster TX16S as a slave to these two txs and that doesn’t work either.

It looks like a firmware issue to me and I can think of nothing else to try to make this work.

3djc, 2021-11-13:
As shown above x7 linked to Tx16s works both ways. What kind of cable did you use. Used a mono one

shaun190, 2021-11-13:
Hi 
My problem is definitely not cable related, I have tried multiple versions of mono, stereo, four channel, three channel and a Radiomaster WT01 wireless link.
None of them work with the newer txs as a master but all of them work with the older txs as a master.  I believe a UK importer of these txs has also tried this and founded the same issue.  Their recommendation is to roll back the firmware but I have been unable to try this as the recent certificate issues on the OpenTx site have gone beyond my computer knowledge to solve.

ChrisOhara57, 2021-11-13:
Hi
only to ensure myself that I didn't make any silly mistake I flashed OpenTx 2.3.14 on my Taranis x9Lite S again today.

Took two cables one mono and one stereo - checked that both work with 2.3.11
Flashed to 2.3.14 - none of the cable worked.
Flashed back to 2.3.11 - both cable work again.

shaun190, 2021-11-13:
Thanks Chrisohara57
Your experience mirrors mine.

I am thinking about edgetx, I wonder if the issue is in this too?

ChrisOhara57, 2021-11-13:
> Thanks Chrisohara57 Your experience mirrors mine.
> 
> I am thinking about edgetx, I wonder if the issue is in this too?

Depends on what you are flying ( edgetx is more a development platform).
I did flash to 2.3.11 and it is working fine and stable.
I even kept the SC-card content of 2.3.14 (Works fine. You have only to edit the file opentx.sdcard)

ChrisOhara57, 2021-11-13:
Sorry, I closed the issue accidently

justtrashmaker, 2021-11-14:
This is where I got my old firmware.

https://www.frsky-rc.com/download/

@3djc 
Can you take a picture of your circuit board? Do you have a multimeter? Can you check the switch on the DSC? With AND without a plug?

shaun190, 2021-11-14:
Thanks for the hint justtrashmaker.

I spent some time this afternoon changing the firmware on one of the QX7’s with 2.3.14 to 2.3.9 which is on the Frsky site and it looks like this has solved the problem. 

I ran out of time in the end but I will have another check of this in the next day or two and post again.

ChrisOhara57, 2021-12-03:
Hi
Meanwhile I have contacted the German dealer (Engel).
He is sure that this is a firmware issue.
He got a QX7 from a customer with the same problem.
He first replaced the dsc socket and then replaced the whole board.
Both actions didn't solve the problem.
But after flashing 2.3.09 everything worked fine.
He also informed Bertrand.
Regards Chris

3djc, 2021-12-03:
x7 access have the hardware inverted. Bertrand is investigating what other radio have undocumented changes on this

ChrisOhara57, 2021-12-03:
Good news.
Thank's for the info.

ChrisOhara57, 2021-12-03:
Does "hardware inverted" mean that the switch pulls the pin to high instead of low (or vice versa)?

If YES
Not knowing details about the hardware and the firmware the following proposal might be silly!

Instead of putting a fix configuration into the code related to a specific hardware version it could be better to check the different possible configurations for a valid signal.
Before a new Master/Slave configuration can be used everyone has to go through the calibration procedure in the radio settings anyway.
While starting this menu the firmware could check whether there is a valid signal. If not it could try an other configuration  and stores the information if it succeeds.

Regards Chris

justtrashmaker, 2021-12-03:
And again ... the problem is in the software. Everything works with the old firmware (2.3.9). Alternatively, you can bridge the contact on the DSC socket. That worked for me too, but the easier way would be to fix the firmware.

3djc, 2021-12-03:
2.3.9 did not check the plug, and therefore created false trainer lost messages. We will not be going back to that. We just need to have FrSky provide proper details about their radios so that proper check can be implemented

shaun190, 2021-12-03:
Interesting to see this develop and pleased to realise it was a genuine problem and not me missing something simple.
 justtrashmaker what did you mean by bridging the contact on the dsc socket please?

ChrisOhara57, 2021-12-10:
Hello
Is there any new information on this issues?
If there is a way I can help or try something please let me know.
(Eventually there is a work around by modifying a cable?)
Any feedback appreciated.
Regards Chris

shaun190, 2021-12-12:
Hi 
I’m also interested in any fix anyone has out there, I still have the problem with two QX7 tx’s running 2.3.14 which will not work as master tx’s.  
If there is a simple cable mod or other change that is possible with the current firmware this would be great.

ChrisOhara57, 2021-12-14:
> 2.3.9 did not check the plug, and therefore created false trainer lost messages. We will not be going back to that. We just need to have FrSky provide proper details about their radios so that proper check can be implemented

3djc, You are right there.

Does "hardware inverted" mean that the switch in the DSC socket is opened instead of closed (or vice versa)?

If YES:  Not knowing details about the hardware and the firmware the following proposal might be silly!

Instead of putting a fix configuration into the code related to a specific hardware version it could be better to check the different possible configurations for a valid signal.
Before a new Master/Jack configuration can be used everyone has to go through the calibration procedure in the radio settings anyway.
While starting this menu the firmware could check whether there is a valid signal. If not it could try the other configuration and stores the information if it succeeds.

Regards Chris

justtrashmaker, 2021-12-16:
The switch on the socket switches the signal (PPM) to another pin of the controller. If this is opened (plug inserted), the signal is missing. I don't know why it was built that way. If you bridge this, the connection remains. However, I don't know what other problems can arise if you leave this bridge in.

shaun190, 2021-12-16:
Thanks for the information.
I only fly conventional RC aircraft using my tx’s, does this hardware issue cause a problem if you fly fpv and use the dsc socket for head tracking goggles or something similar?
Or is this done another way?

ChrisOhara57, 2021-12-22:
> 2.3.9 did not check the plug, and therefore created false trainer lost messages. We will not be going back to that. We just need to have FrSky provide proper details about their radios so that proper check can be implemented

Hello 3djc
If it is a matter of switch "normally opened" instead of  "normally closed" there could be an other workaround.
Why not split the option "Master/Jack" into two options:
       Master/Jack-0     (for open switch)
       Master/Jack-1     (for closed switch)
It is not elegant but at least it could work.

**To be honest I am worried that Frsky might have fitted wrong sockets randomly.**
Regards Chris

3djc, 2021-12-22:
We will not push inconvenance to end users unless we have to. At this point in time, FrSky need to provide data about how their radio is wired. If they tell us they mixed up, then we will look at alternatives, but not until then

dwhacks, 2022-01-03:
I'm also having this issue with my x9 lite, and it took me a lot of searching on the web to even find this. 
If I don't quite push the plug in the jack it seems to work, but obviously this isn't a great solution.

Hopefully there's a fix for this soon, as my wife has finally agreed to learn how to fly, ha.

closed 2022-01-06 by bsongis

anthropo, 2022-02-19:
> We will not push inconvenance to end users unless we have to. At this point in time, FrSky need to provide data about how their radio is wired. If they tell us they mixed up, then we will look at alternatives, but not until then

Is there a chance that fsKy answers or will we stay with a non working master mode on our Q X7 ?

ChrisOhara57, 2022-02-19:
This bug has been solved with #8811 for x9lite and QX7
Should work with the nightlys (but I haven't tried myself)

We can only hope that the 2.3.15 will be released in the near future.

commit ba221850 (2022-01-06): X9 Lite Trainer detect inverted - Fixes #8670