opentx/opentx #8671: Support for Hebrew language

i3dm, 2021-08-19:
Please add HEBREW language support to OpenTX. 
i can offer help in translating the text files if you tell me how i can do it.
thanks.

3djc, 2021-08-19:
If you are talking companion, then have a look here ; https://github.com/opentx/opentx/wiki/Flow-for-the-Translators

If you are thinking about the radio, then it is much much more complicated, and can only be achieved in 2.4, and only on colorlcd radios (if possible at all, unsure about the right to left implications)

i3dm, 2021-08-19:
i meant the radio.
lets try it left to right and see? it may be useable

lidorshimoni, 2022-06-29:
Hi, I'm currently looking into implementing it. would be glad to have any suggestions or tips that might help.