opentx/opentx #8672: Booting up RadioMaster TS16S hangs up.

CarlouT, 2021-08-19:
I just bought a Radiomaster TS16S. I was able to use it for a while to setup but lately, I am not able to boot the radio. When I push the power button, the splash screen (Welcome screen) appears but it will stay there and does nothing. Pressing the power button again to turn it off does nothing as well. Which means the radio has hang up. I think it has entered an endless loop somewhere in its programs. To power it off, I have to remove the battery. Putting the battery back and repeatedly doing the same steps, sometimes the radio will boot in Emergency Mode. In this mode, the radio can be turned off with the power button. if you are lucky, sometimes the radio will boot properly which is very annoying.

I've been researching all over the place but recommendations I found does not convince me. Until I remembered, I replaced the SD card with a bigger capacity. I used a size that is recommended but have reused an old SD card I have. Just for curiosity, I copied the contents of this SD card to the original SD card that came with the radio and wah lah.... the issue was fixed. I'm back to business. Which means TS16S has some issues reading my old SD card even if it can be read in a PC properly.
Should anybody experience the same, try changing the SD card before trying the recommendations found from Google. One of the recommendations is if you have replaced the splash picture, the size of the picture you have used is not the correct size causing the issue. Yes I have replaced the splash screen. I tried restoring it to the original picture, but this did not fix the issue. Another recommendation is one of the ribbon wires inside has some loose connection. I did not open my radio as it is very new, and I don't believe this is the root cause.

To the developers of Open TX, What does Emergency Mode mean? When and why will the radio go to this mode. Kindly enlighten us and what do we do when the radio is in this mode.
[image]

CarlouT, 2021-08-19:
Interaction from the Public on this issue here. https://www.facebook.com/groups/827280541160158/permalink/943994982822046/?comment_id=944003546154523&reply_comment_id=944028499485361&notif_id=1629408555367160&notif_t=group_comment&ref=notif

CarlouT, 2021-08-20:
I guess for you guys to test this, look for a crappy SD card. Copy the necessary files and then plug it in the radio. All I did is to replace the old SD Card that I used by copying the contents of this card to the original SD card that came with the unit. And the issue went away. So I can confirm that the root cause of the issue is the SD card.

CarlouT, 2021-08-20:
By the way, here's the old SD card that use. One year old. Its a reputable card.

[image]

3djc, 2021-08-20:
Clearly SD related, not OpenTX related

closed 2021-08-20 by 3djc