opentx/opentx #8676: Incorrect Battery Offset Displayed in Companion

Hartinger, 2021-08-21:
I have calibrated the display of the battery voltage in my Horus X10S Express with a precise multimeter. To match the voltage display of the Radio with the value indicated on the multimeter, I had to increase the voltage value by 40mV. When I then imported the OTX file into Companion, a battery offset of 0,4V was displayed on the hardware page. Not a big issue but I thought it's worth mentioning.

elecpower, 2021-10-09:
The value is actually the number of voltage steps (+/-127) offset from the metered voltage. The voltage each step represents varies from radio to radio. Rather than display erroneous data which should be read only anyway, better to remove and leave it to the radio.

Hartinger, 2021-10-09:
Good decision. 👍