opentx/opentx #8678: Taranis X-Lite Pot S1 Companion vs Tx

KA8500, 2021-08-22:
**Output of Pot S1 in companion do not correspond with the Transmitter of Taranis X-Lite:**
If you move Pot S1 and S2 in the companion from the left detent clockwise to the right detent, the bars in chanel monitor will move from left to right and the value will go from -100% to +100%. This seems to be correct as the S1 and S2 from the Taranis X9D work same. Now, you take the transmitter X-Lite and move S1 an S2: While S2 is working as expected, S1 values go from +100% to -100% and the bar from right to left, which is opposite of what you will expect according companion and opposite of S2. With the Taranis X9D+ all correct. My conclusion is that companion is correct and the wiring of the X-Lite S1 pot ist mixed up. 
It looks like a FrSky hardware bug. On the other hand the companion should look like the real Tx. In this case you have to alter the turning direction of S1 in the companion. This means S1 of the X-Lite and X9D work in different directions to get the same output. 
**Solution:** 
Wiring of pot S1 is wrong and should be changed by the manufacturer. If not possible, make the same "mistake" in companion. Than S1 in companion and X-Lite will work same direction, but different than X9D.

 - OpenTX version: all (2.3.14 and older)
 - Hardware Taranis X-Lite Pro and X9D+
 - OpenTX Companion: all (2.3.14 and older)

regards 
Kurt

3djc, 2021-08-22:
Companion pots and sliders do NOT intend to map physical behavior of radio.

Also, on xlite, they are not pots (that are all CW), but sliders, that go to 100% toward top/center

closed 2021-08-22 by 3djc

KA8500, 2021-08-22:
I am not satisfied with your answer:
_Companion pots and sliders do NOT intend to map physical behavior of radio._ Ok, but till now this is the only pot and slider in my equipment which do not map physical behavior of radio.

_Also, on xlite, they are not pots (that are all CW), but sliders, that go to 100% toward top/center_ This is not true. If pot S1 and S2 would be sliders, they have to go plus 100% toward the center. This ist not true in companion (both go CW to plus 100%, as pots do) and also not true in the radio as both "sliders" S1 and S2 go to minus 100% when moving to center. 

Thank you for the answer. I am now sure that companion will not be changed. I will change the wiring of S1 to have a perfect solution. 

Regards 
Kurt