opentx/opentx #8688: Delay Parameter for Autooffset of Telemetry (see also issue #5787)

ChrisOhara57, 2021-08-29:
Problem see also  closed issue #5787

I wrote a script (Training App for RES Flying).
      https://repository.justfly.solutions/index.php?view=product&id=119:res-trainings-app
This script does recognize the start of the plane when altitude reaches 5m (landing 2m altitude)

I have installed an Archer GR6 (also flashed lasted firmware to it).
Our location is about 380m above sea level.
When the receiver is switched on it immediately starts sending with an altitude of Zero and then the values increases to 380m in about 3 seconds.
Autooffset does occur after roundabout 2 seconds what is at about 250m altitude.
But altitude still rises another 130 meter in one second.
This results in a groundlevel Alt of -130m   and   VSpd+ does shows about 130m/s

Problem now is that the script "thinks" the plane has started but it will never recognize a landing because that would have to be  130m below real ground level.
In fact i can handle that in my script with some extra coding. (always checking status of RSSI and resetting 3s after switch on)
And for normal usage you have to remember to do a manual reset after a few seconds.

But the discussions of that issue shows that it is a nuisance to others as well. (Other receivers like GR8 are affected also.)
https://frsky-forum.de/index.php?thread/4997-vario-automatisch-zur%C3%BCcksetzen-sowie-senderspannung-anzeigen-lassen/&pageNo=1

**Proposal:
Install a Parameter that can be defined by the User either as a parameter of the Modell or of the whole Transmitter.
This parameter defines how many seconds the autooffset function is delayed. (Default value could be 3 or 5 seconds)
This would be a general easy to understand solution.**

3djc, 2021-08-29:
Or ask FrSky to fix the root issue?

kilrah, 2021-08-29:
You can already do it yourself by using the Reset (telemetry/sensor) special function with a logical switch that has a delay set.

ChrisOhara57, 2021-08-29:
Hello 
Thank you for the fast response.
I've addressed that problem to Frsky. Did get a nice mail back that they will ask the firmware guy whether he is aware of the problem -  no other response. (I know I was not the first one to ask.)

I also know that openXsensor does it better with the baro. 
But then on the other hand  my openXsensor Gyro (MPU6050) does send false data for nearly one minute.

So I believe there will always be a sensor or a receiver or a firmware version that will have this problem.

For the solution with logical switch. I took an example from the EngelMT forum (one guy did it with the help of a few experienced users. If it´s really that complicated then this is no solution for the average user. 
[image]

For the problem with the  lua script this logical switch would not help either. The function would already have started when the reset occurs.

**But I have to confess that I haven't thought of that before because my proposal as it is would not help for the script either.
The solution for that would only be to suppress all (or at least most of the) telemetry data for the first x seconds.**

While there is a workaround for this issue I do not mind if the issue is closed.

You do a great job, Thank you for the effort.
Regards ChrisOHara

kilrah, 2021-08-30:
You don't need all that stuff, just one LS that's always true with the duration you want as delay and a Reset SF either on the whole telemetry or just the sensor.

[image]
[image]

ChrisOhara57, 2021-08-30:
Hello
Does not do the trick under 2.3.7
As soon as I set the duration to something other than "---" like "5" the logical switch does become "false/off".
When I leave it "---" then L01 becomes "true/on"  but the reset does happen without time limit (infinitely).
If I concentrate I can also see a flickering of 0.0 of the Alt value in telemetry screen.

I do trust your word that this works but it does not on my taranis under oTx 2.3.7
And also it does not seem to be straight forward (logic)  to me.
I would never have tried something like that.
When I switch the receiver off and on again why should there be a reset once more?

But please leave it as it is.
I do have my own solution (I also attached a sheet with the logical switches I use)

Thanks a lot
ChrisOhara

PS: By the way the term  "a<x"    with "RSSSI < 2"   is always "false/off" no matter if the receiver is switched on or off.
      So you have to work  with  "Not   RSSI > 2" instead
      (oTx 2.3.7)

[image]
[image]
[image]