opentx/opentx #8689: External module selection

FPVxfun, 2021-08-30:
It is not possible to reduce the number of channel in External RF -> Mode DSM2 -> DSMX

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Model Setup'
2. Click on 'External RF'
3. Select 'DSM 2' protocol
4. Select 'DSMX' sub-protocol
5. CH1-CH8 could not be reduced to CH1-CH6 for 6 channel receivers. In this way the bind could not take place

**Expected behavior**
If possible, please give us the possibility of reducing channel as per Internal Module selection

**General information**
 - OpenTX version (e.g. 2.3.14)
 - Hardware (Radiomaster TX16S)
 - OpenTX Companion (e.g. 2.3.14 on Win10 64)

3djc, 2021-08-30:
That’s not how it works. If you have issues binding dsm, please refer to : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6
And to this thread : https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1509

closed 2021-08-30 by 3djc

FPVxfun, 2021-08-30:
@3djc all the setting you provided, are for INTERNAL MULTIPROTOCOL RF MODULE. But I need the same for EXTERNAL RF MODULE (I'm using an original Spektrum DX5e external module)

kilrah, 2021-08-31:
The number of channels is hardcoded to 6, the 2nd number doesn't actually mean anything for the DSM2 proto.