opentx/opentx #8690: Script Syntax Error after update to 2.3.14 with 9+SE (2019)

Samedwin, 2021-09-01:
I am trying to run Lua to enable CRSF on external XJT port. I have the Crossfire module for my taranis. I have the 9+ SE (2019) Tx. 

I just did a fresh flash of 2.3.14 (previously using version 2.3.3). New Taranis Tx did not have SD card. I updated the SD card with current SD card release 2.3.0036. I used the firmware file - x9d+2019_en_210208.dfu (bin).
I did not use the Companion - just did all updates via SD cars/Computer.
So-
On the Tx, Upon Opening the files  "SCRIPTS", "TOOLS", "CROSSFIRE" and executing the file crossfire.lua. I get the following error. 
"
Script syntax error
TOOLS/CROSSFIRE/crossfire.lua:62
attempt to call global 'crossf
"
For some reason the SD card contents contains the crossfire.lua script in multiple places. So, I thought maybe one was corrupted, I'll try another one.
On my second effort to execute get the CRSF external module- I tried the crossfire.lua file that's under "SCRIPTS" "TOOLS" , upon execution I get a similar error but slightly different. This error reads 
"
Script syntax error
?:0
attempt tp call field 'crossfiretelemetryPo
"
looks like there was more to the error but it spilled off the screen.

TO REPRODUCE. I did a fresh flash of the software, same version. I also changed the SD Card out to a new one, re-downloaded the SD card contents, and put on the new card, did a new flash of the software version - x9d+2019_en_210208.dfu (bin)

I was of course expecting to get the CRSF selection in the External Module menu. Obviously I checked and I do not get a CRSF option for external module. So the lua did not work.
If you really want screen shots I can provide, but they say exactly what I stated above.

**General information**
 - OpenTX version (2.3.14)
 - Hardware Frsky X9D+ SE 2019
 - Did not use OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)
 - Using Windows Surface machine 64 bit 
 - Used bot a 1 GB and 32 Gb freshly formatted FAT32 SD cards
 
Any help with these issues would be greatly appreciated. I obviously have scoured the errors and cannot find these errors. I am not sure what to do. I upgraded from an old taranis to the 2019. My old Taranis worked great but I just wanted to get the new one. My luck with software/tech! Bad luck if any ! lol

3djc, 2021-09-01:
Have you set baudrate to 115k in radio/hardware ?

kilrah, 2021-09-01:
Did you check the version page after flashing 2.3.14 to make sure that it really flashed? Flash it from bootloader after putting the file on the card.