opentx/opentx #8693: No possibility to work with Internal XJT HORUS S12 and external Multimodul together ????

Githubbachelor, 2021-09-03:
Hallo dear Specialists
I have another time a Firmware latest which does not support longer my 6 Airplanes with installed 2 Receiver!

Before, time ago you fixed it to use both receiver, the internal and external MULTI Modul. I want to use my planes in Holiday now and again that desaster that the use of Intern and extern Multimodul in my Horus is not possible??? If I want to use the extern MULTI M. the internal will be disabled??? Did we go backwards or forwards? I am sitting now with my big planes at a lake and can do nothing!! Why can I not use longer my both receiver Internal D8 with channel 1-8 and external over my Multi Modul latest firmware another D8 with channel 9-16? That was working all the years! Maybe you can tell me with which firmware for Horus S12 it does work.
Hope for a quick reply to use my planes at the lake now and not next year. And I think that it is not a problem with the MULTI MODUL, I try also older firmware for Multi M. with same problems.

Githubbachelor, 2021-09-03:
[image]

Githubbachelor, 2021-09-03:
I use two Receiver D8 with DSMX or D8 with D8 and now all not possible again after the last bug I reclamed - there was the problem that the the binding was losing after a new switch on of the Horus. And you fixed that problem. But not we cannot bind two receiver anymore!

Githubbachelor, 2021-10-01:
There was a version which work with both tx moduls, internal XJT and Extern Muli all protokols with X12S - now again not working! I use many models with d8 at internal and different receiver for external Modul - please fix that bug

Githubbachelor, 2021-10-06:
Why after a lot month not solved? in any versions before? can you tell me the latest version which was working with 2 receiver? i will downgrade

raphaelcoeffic, 2021-10-06:
> Why after a lot month not solved? in any versions before? can you tell me the latest version which was working with 2 receiver? i will downgrade

You do know that both modules are using the S.PORT line, right? So the point is more, how to convince these modules they should not fight for the same telemetry line. AFAIK, the multimodule does not have such a mode. Your internal XJT does not have this either.

Githubbachelor, 2021-10-06:
So what is possible to solve that? Mayb there is a Version where only the
internal XJT has Telemetry. I do not need Telemetry in the external
Multimodul. I need the external for Lights and functions. The internal only
for to fly.

S12 and external Multimodul together ???? (#8693)

 

<https://github.com/opentx/opentx/issues/8693#issuecomment-936642520> , or
unsubscribe
R3E3ANCNFSM5DLUBVRQ> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AFMJDOSZEWRWXFW2Q5Y5RWLUFR3E3A5CNFS
M5DLUBVR2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOG7KAPWA
.gif>

raphaelcoeffic, 2021-10-07:
@Githubbachelor whether or not the MPM is able to leave the S.PORT line alone is something you should ask to the MPM project. This is not part of OpenTx.

Githubbachelor, 2021-10-11:
Is there a way to fix it in the Multi Module? I only need telemetry in the Horus 12 internal XJT Module. No need telemetry in the Multimodal as a second transmitter together with internal.

3djc, 2022-01-21:
It does have nothing to do with development, it is simply an hardware limitation. On radio that do not have this limitation, it is possible

raphaelcoeffic, 2022-01-22:
> It does have nothing to do with development, it is simply an hardware limitation. On radio that do not have this limitation, it is possible

> Horus 12 internal XJT Module

I second @3djc. The issue is  the following: the S.PORT line is shared by the internal XJT and Multimodule. Both modules use it as their telemetry link, so that some synchronisation would be required, which would need changes to both firmwares of both modules.

Also, the S.PORT protocol implemented by the internal module is a master/slave bus. It does not support multiple masters (read XJT and MPM here). So I doubt this would ever work.

There is a possibility at the hardware level however that implies using a "Upgrade ISRM module" as the internal one, with a small hardware modification: the S.PORT needs to be separated from the ISRM, so that it cannot block it. Please note that the ISRM does not need or require the S.PORT line for anything.