opentx/opentx #8696: Multiple Logs with Taranis X-Lite Pro and Renaming Logs are impossible

AlphaCharlyX, 2021-09-10:
General information
    OpenTX version 2.3.14 or 1e09791a)
    Hardware Frsky Taranis X-Lite Pro
    OpenTX Companion 2.3.14 on Win10

Hello,
with the Taranis X-Lite Pro and OpenTX 2.3.14 it is impossible to record more than one Log at the same day with the same model.
So you aren't able to take multiple Logs next to each other with one model at one day.

I think this must be a "File-Naming"-issue.

Renaming the Log-Files on the SD-card through the Taranis X-Lite Pro is impossible as well. I thought this could be a workaround...

Greetings Andreas

3djc, 2021-09-10:
There is one log per day that includes all the flights for that day

AlphaCharlyX, 2021-09-10:
Thank you very much. That I didn#t know

closed 2021-09-10 by AlphaCharlyX