opentx/opentx #8699: OpenTX 2.3.14 Linux Version - strange simulator switch behavior at start

nostromo123, 2021-09-12:
This is about a strange behavior of switches in simulator at start - openTX 2.3.14 'deb Version'.

I made screen recordings the simulator of an arduplane model.

Switches SA and SB -> flightmodes, SD -> Lipo voltage and RRSI value and SH -> arming.

The simulator with that model in openTX 2.2.4 'deb'.

https://www.youtube.com/watch?v=RDV2ZfviYhw

Behavior as expected!

and same model with identical settings in openTX 2.3.14 'deb'.

https://www.youtube.com/watch?v=hUJ1n5h6WfM

Have to go through switch position SH, SD and SA/SB to get a 'normal' operation.
(At start, I get 'armed' , Lipo voltage and NOT manual mode, which is not correct)

Copied the model 'otx file' to a Windows10 installation of openTX 2.3.14 and there that simulator does not show this issues. 

br Karl

elecpower, 2021-09-12:
@nathantsoi the two videos show two different radios being simulated. Also the simulator had a major rewrite for 2.3. Can you please post a video of same radio and model for 2.3.14 Win 10.

3djc, 2021-09-12:
There is also the feature of switches start position possibly have an impact ?

elecpower, 2021-09-12:
I did think of that possibility too but in theory they should be set to the saved positions if that option is selected. 
@nostromo123 can you please check your simulator settings as they are saved per OS and ensure they are the same across OSes.
[image]

nostromo123, 2021-09-12:
@elecpower 

Yes, two different radios. My X9D on 2.2.4 and my X10S Express on 2.3.14. But upgrading my good running X9D is a thing, I would not like to do. 'Never change .....'

I am here with DEBIAN Bullseye and WIN 10.

I have attached that arduplane model setting printout for Linux '224.pdf' (X9D) and Linux '2314.pdf' (X10S Express) and '2314_WIN10' (X10S Express)

Also see the screenshots of both sim settings in Linux (224 and 2314)

By the way, this only affects the simulator, on the radio (X10S Express), behavior is normal.
And putting those 3 switches (SA, SB, SD and SH) through once resolves this issue.

Maybe there are faults in my programming that model, but then why only on the 2.3.14 Linux sim??? 

br Karl

[224.pdf](https://github.com/opentx/opentx/files/7149654/224.pdf)
[2314.pdf](https://github.com/opentx/opentx/files/7149655/2314.pdf)
[2314_WIN10.pdf](https://github.com/opentx/opentx/files/7149657/2314_WIN10.pdf)

[image]

[image]

elecpower, 2021-09-12:
@nostromo123 thank you for the information. Appears it could be an OS quirk but to be sure could you please upload the troublesome otx so I can run up in my Linux environment.

nostromo123, 2021-09-12:
@elecpower 

No problem, please find otx attached. (Replaced .otx with .zip for upload)

br Karl

[document1.zip](https://github.com/opentx/opentx/files/7150267/document1.zip)

elecpower, 2021-09-18:
Cannot reproduce using your model file on Ubuntu 20.04

nostromo123, 2021-09-18:
Thank you for testing!

So I installed Xubuntu 20.04 in a VM here.

Installed openTX 2.3.14 and loaded the SD card content and added the AMBER sounds.

Configured openTX and loaded that otx file.

This is what I get

[image]

Switches (I use) SA, SB, SD and SH in up position.

I don't hear 'throttle disabled', which I should, but I hear 'zero volt', which I should not hear.

And the flight mode is displayed as 'RTL', should be 'manual'. 

So you have more luck then me. ;-)

**But STOP**, then I restarted Xubuntu 20.04, startet opentx, loaded the otx file, started the sim ...... and behavior was as expected and ok.

Second try now with Xubuntu 21.04 in a VM.

First start I have this issue again, restart and it is ok.

What can I say? I have to pin this to DEBIAN. And I have DEBIAN Bullseye and Testing running on three machines here, two desktop pcs and one thinkpad .....

br Karl

nostromo123, 2021-09-20:
Next try on that.

Build opentx in a DEBIAN Bullseye VM. Took me some hours ....

[image]

First start issues, second start all is fine - like on the Buntus here.

I give up, but one question keeps in my mind. Who build and generated that 'companion23_2.3.14_amd64.deb' file on what system?

br KH

closed 2021-09-20 by nostromo123