opentx/opentx #8701: a<x in logical conditions for channels (ie CH8<0) does not work properly.

skorokithakis, 2021-09-16:
**Describe the bug**
I'm adding a logical condition to check for when CH<0, and it does not change. I think it has to do with the fact that that channel's output only comes from special function overrides.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a switch override to set CH8 to 100.
2. Add a logical condition for when CH8<0.
3. Logical condition never changes state, no matter the switch position.

**Expected behavior**
I expect that the override will feed into the logical condition and change its state when the channel output is different.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version: 2.3V0034
 - OpenTX Companion simulator
 - OpenTX Companion 2.3.10-otx on Ubuntu 20.04

3djc, 2021-09-16:
Working as designed

closed 2021-09-16 by 3djc

skorokithakis, 2021-09-16:
Why is the design that the channel's output will be something different from the channel's output?

3djc, 2021-09-16:
Overrides doesn't affect mixer chains, doesn't obey limits or anything. As said earlier, if you do not like this behaviour, use a mixer line to perform the same role, in a way that you seem to like more

skorokithakis, 2021-09-16:
Sure, but then shouldn't "Channel output" be renamed to something else? When I read "channel output" I think "what the channel outputs", and not "what it would output if it didn't output what it does".

kilrah, 2021-09-16:
CHx are the outputs of each channel as on the mixer page, not the servo outputs sent out to the module. You'll see that anything on the Outputs page isn't taken into account either.

skorokithakis, 2021-09-16:
Hmm, I see, thank you.

timtuxworth, 2021-09-16:
I can kind of see Stavros point. How about this?

On the outputs page - if one of the channels has an override in Special Functions can there be maybe a flag or something that warns you that there is an override? Remember that overrides are often conditional, For example I use a special function for my throttle cut. If Switch A is on - then override Channel 3 to -100. So if switch A is off there will not be an override and the output will be “the output”.

So - how about a marker on the output line to say “possible override exists”?

notlebowski, 2021-09-17:
The whole "channel override" is an anomaly and exists upon requests. I'm sure the devs wouldn't have implemented it they knew the amount of debate and confusion it would bring in the future.

In 99 of the 100 use cases there is a better way than "override" to influence that channel.
IMHO you are always better of to use the mixer page to override the channel.
If you know you're not using "override", you will see everything happening to that channel in 1 page.

Now you want some kind of flag to tell you the output can not be trusted.

You should only use override if
-  you deliberately don't want the other logic in your setup to "know" the state of what the channel is doing
- Tu want to send values to the servo which are not influenced by the servo range
- you may feel insecure about your programming and you want to be absolutely sure the servo is getting a certain value (in case of throttle override).

I'm using a := (replace) mixer line with MAX as the last one for a throttle override. This means that -100 in a normal setup.

The current behaviour of the CH-value not reacting on the overrides is something that can **NOT** be changed as it is used by some people intentionally.

skorokithakis, 2021-09-17:
The good thing about overrides is that they support precedence. If I have three switches all setting a channel to some value, I can say which one should win. That doesn't happen with mixers, does it?

notlebowski, 2021-09-17:
In mixers you can always have a replace line which overrides every line that becomes before it...
Much less blunt than an override.

timtuxworth, 2021-09-17:
Thanks @notlebowski - I appreciate the detailed explanation. I'll change how I do throttle cut.

skorokithakis, 2021-09-17:
Hmm, yeah, I think I'm swayed as well, thanks.

notlebowski, 2021-09-17:
TBH

People who advocate the use of "Override Channel" can come up with arguments to use it for a throttle cut. I personally think they are a bit far fetched if you know what you are doing.

Because the Outputs page will have "the penultimate say" about what is sent to the servo's this means that a "reverse channel" on your throttle will put oyur motor into full speed. Also changing the range will influence it.

I personally think this argument should be placed at the "uncertain" bullet in the short list I gave.

I do a lot of "reading the channels" instead of reading the inputs that control the channels when I write setups in OpenTX. This way I can be more certain of the outcome in my setups.
That's why I never use override channels.

timtuxworth, 2021-09-17:
It does make me think  it would be "nice" to have way to show the logic for each output as a diagram. Perhaps in the simulator? Even better - an animated diagram that hightlights the logic path driving the current output and changes as sticks move or switches are changed.
It would look awesome and be very, very helpful when figuring out what is going on with a particular setup. 
I know it's easy to dream of these things, but a lot of work to implement.

timtuxworth, 2021-09-17:
I'm starting to get it, but it wasn't easy when I was getting started. Think of it as a way to show new users or occasional users what is going on in _your_ head. :-)