opentx/opentx #8703: OpenTX Companion 2.3.14 - The Scale setting only allow integers while a radio allows up to 2 decimal places.

Miami-Mike, 2021-09-17:
Suppose, for example, that you want to set up sensor RxBt as an input with a Scale setting of 5.2, so that your fully-charged model battery, at 5.2 volts, shows as 100%, and the voltage of a partially depleted battery shows as a percentage of that fully-charged voltage.

If you try to use Companion to enter this fractional value it will be changed to an integer when you click "OK" to save it. The value of 5.2 in this example will be changed to 5.0. However, if you enter the value with a radio or with Companion's radio simulator, the value you enter will be accepted and will work properly. A telemetry value of 5.2 volts will display as 100% and a value of 5.0 volts will display as about 96% - 97%.

And if you import that properly-working setup into Companion and open the editing window for that input, you'll see 5.0 but in the simulator the input will still work as if the scale were set to 5.2. That is, unless you click "OK" in the input editing window, in which case the 5.2 value will get changed to 5.0.     

If you change the sensor's precision to "0.00" then with a radio or the radio simulator you can set the scale to a precision of 0.01, but Companion will still change the value to an integer.

elecpower, 2021-09-18:
confirmed

Miami-Mike, 2021-09-18:
An additional note: Companion 2.1.9 has this same problem, so perhaps Companion has always behaved this way.

PR #8705 (2021-09-19): Companion inputs scale fix and enhancements
Fixes #8703 
Add sensor unit to edit dialog
Add non-zero scale to inputs list and model print and compare
closed 2021-09-19 by 3djc