opentx/opentx #8704: error in lua script "FrSky SBEC" to set Neuron bec voltage

strgaltdel, 2021-09-18:
Hi,

this problem still exists since several otx revisions:
Under certain circumstances the "BEC lua script" will crash.

[image]

root cause is the existence of calculated sensors, like this:

[image]

in case you delete this sensor the script will run.
Easy solution is to change line 259 into "if sensor.id ~= nil ..."

[image]

imho another "problem" could occur when bec sensor is placed in an "telemetry slot" > 40,
it won't be detected by the script.
So i would recommend to increase index loop up to the limit of 64 sensors

regards
Udo

3djc, 2021-09-18:
REally FrSky doing since they own this script, but we might fix it nontheless, thanks for that

kiwi41000, 2021-11-26:
I’ve been working on this problem for what seems like forever. I find this bit of information I ended up deleting all the sensors rediscovering  them and the BEC adjustment works perfectly ,thank you for this information. Kiwi4

strgaltdel, 2022-06-13:
Ich kehre zurück am 27.06.2022.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] error in lua
script "FrSky SBEC" to set Neuron bec voltage (#8704)" gesendet am 13.06.2022 01:09:01.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.