opentx/opentx #8708: Multi-Protocoll Module JP4IN1-SE with FrSky X10S Express and OpenTX version 2.3.14 doesn't work

AlphaCharlyX, 2021-09-20:
General information
OpenTX version 2.3.14 or 1e09791a)
Hardware Frsky X10S Express
FW: opentx-x10express

Hardware Frsky Taranis X-Lite Pro

OpenTX Companion 2.3.14 on Win10

Hello,
I have the problem, that my new MultiProtocoll-Module JP4IN1-SE doesn't work with my FrSky X10S Express and OpenTX version 2.3.14.

As you can see on the picture on the radio and opentx the status of the Multimodule is "**No Multi-Telemetry detected**". Binding models and Flashing the Multimodule is so impossible. 

On my Taranis X9d and on my X-lite Pro with JR-adapter the Multimodule works perfectly (binding models and flashing the module with the newest firmware is no problem here.

On the Multimodule itself is the newest Firmware 1.3.3.0 successfully installed.

Thank you for your Help and openTX.

[image]

kilrah, 2021-09-20:
You are most likely using the wrong firmware on the module, pay attention to the modules that come with the wrong chip version andneed specific firmware.

closed 2021-09-20 by kilrah

AlphaCharlyX, 2021-09-21:
Hello, thank you for your fast answer. Did you check this hardware configuration with opentx 2.3.14? I am pretty sure that i have the right firmware on the Multimodule because the same module works perfectly on my xlite pro and on my x9d 2019.

AlphaCharlyX, 2021-10-03:
The Problem has been solved here. Thanks to Pascall and the people helped here.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621