opentx/opentx #8710: widget timer2 does not show days

offer-shmuely, 2021-09-21:
**To Reproduce**
1. set persitent timer 
2. run the timer for more than 24 hours
3. display the timer with widget "timer2"

widget display:
39:41

**Expected behavior**
1d 15:41

[image]

[image]

3djc, 2021-09-21:
We got a request to actually make it display day...

offer-shmuely, 2021-09-21:
now it does.
if the time is below 1 hours --> it will display mm:ss
if the time is above 1 hours but below 1 day --> it will display hh:mm:ss
if the time is above 1 day --> it will display dd hh:mm:ss

Note: I use the timer3 for recrding total fllight time of the model, so it reach high values

MlCHEL1234, 2021-09-21:
Unbelievable - another example of how OpenTX implements sensible user suggestions.

No other system will be able to keep up with this.

kilrah, 2021-09-21:
I prefer seeing it in hours and no doubt others will, so maybe make it a widget option to enable the "day" display.

offer-shmuely, 2021-09-21:
added option "use_days"

closed 2022-10-18 by offer-shmuely