opentx/opentx #8722: Having the throttle above fully idle permanently sets that position when going into calibrate

twostage, 2021-10-01:
Before describing the bug, can I say that openTX is brilliant and a big thank you to the developers.

**Describe the bug**
I went into calibrate on an Radiomaster TX16S and pressed enter without realising the throttle was in the middle position. I moved the stick up and down and spotted it wasn't going below the initial position so I immediately pressed return to get out but that position became set as the lowest throttle position. When I go back into calibrate with the throttle in the idle position it shows it in the middle position and will not calibrate back down. I now only have 0 to 100 instead of -100 to 100 on the throttle.

I've tried re-flashing the TX  and deleting radio.bin but the calibration settings must be saved somewhere.

**To Reproduce**

Put the throttle in the mid position, go into calibrate, press enter and move the throttle. It will now not go lower than the initial position.

**Expected behavior**

Calibrating should start fresh each time and not be bound by previous calibrations. Or there should be some way of clearing a previous calibration manually.

**Screenshots**
Shows throttle in lowest position but calibrate showing it in the middle.

[image]

**General information**
 - OpenTX version 2.3.14
 - Hardware Radiomaster TX16S with Hall Gimbals
 
Thanks for any help.

Jed.

3djc, 2021-10-01:
Not sure what you do wrong but this is not the case, you actually SHOULD start with throttle mid position when you calibrate

closed 2021-10-01 by 3djc

twostage, 2021-10-01:
You're right :-) I was doing it completely wrong. I was trying to start with the on-screen gimbals centered rather than the physical gimbals centered. All OK now. Thanks for your help.