opentx/opentx #8723: Companion update check failed, new version information not found

derpolygraf, 2021-10-01:
Hello, I have a issue with the open tx companion. After I start the program i get a error message: Companion update check failed, new version information not found, what should I do? I am using Win 10 and the newest companion version. Please help!

LG

kilrah, 2021-10-01:
It's being solved right now, you will have to manually redownload and reinstall companion. Or just disable auto update check.

derpolygraf, 2021-10-01:
@kilrah thx for your answer, I reinstalled companion and disabled the automatic updat check up, but iam sitll not able to download a new Firmware: "Firmware update check failed, new version information not found or invalid."

3djc, 2021-10-01:
Yes problem is known and linked to https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/ A new repackaged version of Companion is going to be available soon

kilrah, 2021-10-01:
I said it was being solved, not that it was ready yet... It now is.

http://www.open-tx.org/2021/10/01/companion-downloads

closed 2021-10-01 by kilrah

mestafpv, 2021-10-01:
i think that not solved
[image]

kilrah, 2021-10-01:
If you had just downloaded it before it's likely your browser has cached the old file. Clear your browser cache.

3djc, 2021-10-01:
You can check with the about window

[image]

The new version says Oct 1st

mestafpv, 2021-10-01:
thanks! 

> Si lo acababa de descargar antes, es probable que su navegador haya almacenado en caché el archivo anterior. Limpia la memoria caché de tu navegador.

thanks i will try :)

Chappie6603, 2021-10-01:
Worked like a charm, definitely clear browser cache before downloading!!

Thanks for the fix!

Chappie6603, 2021-10-17:
> Hi
> Just received my Radiomaster tx16s running opentx 2.3.11.
> Checked my root certificate at the site from your link and got a valid page.
> Cleared my firefox cache
> Downloaded and installed companion 2.3.14, for windows 10
> still get the error messages for companion and opentx.
> Don;t know what else to try...
> Nick

Not sure about Firefox, I used edge, but should be the same.
Clear cache first, download new companion 2..3.14, install, and start.

Mike C.Canada

kilrah, 2021-10-17:
Usually that's when people messed up their installs and have companion installed multiple times in different folders.

etskipsey, 2021-10-17:
Hi, Same problem, uninstalled companion, ensuring all bits were gone, reinstalled 2.3.9  (windows 10) same problems .  Suggestions?  Edgar

3djc, 2021-10-17:
ONLY version 2.3.14 dated Oct 1st will work, any older version will not, as the changes have nothing to do with companion per say, but overall ssl certification

etskipsey, 2021-10-17:
Thanks, saves wasting my time, will update to 2.3.14  Help appreciated  Edgar

Taildragger1589, 2021-10-18:
for those who've just about given up...

I finally got mine working after trying over and over with companion 2.3.14.
I cranked up file explorer and deleted everything I could find from opentx.
(the uninstall left a directory in programs 86 and 2 library files in c:/)
then I used the Edge browser instead of firefox to download the companion installer (after clearing the browser cache files)
After installing... it worked fine.
I now have updated firmware and SD card on my Tx16s.