opentx/opentx #8729: Horus X12 Bug "Master/Jack"

matkovits, 2021-10-04:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

matkovits, 2021-10-04:
Hello to All
Max OTX 2.3.11
My Hardware: Horus X12, Radiomaster TX16S

Horus X12S Master/Jack problem

1. I now have a R8SB receiver lying here, so you can set output 8 as a PPM output when binding. Then the Horus bound with this recipient. The jack plug is plugged into the Radiomaster and the RM is set as the teacher transmitter. In the RM's Sys menu you can see that the Horus is sending a student signal to the RM via the receiver. (the cal values change continuously)

2. The experimental setup is the other way around, ie the RM is bound to the receiver and the jack cable is plugged into the Horus (Horus as teacher transmitter) and the Horus in the Sys menu shows the cal values 0.0. ie no signal input. Just like with a direct cable connection.

It seems to have a bug with the Horus.

The jack socket of the Horus delivers an output student signal, but no student input is established.

regards from Austria

matkovits, 2021-10-04:
.... and how can I determine or fix it

Target0815, 2021-10-04:
@3djc : No ... @matkovits can use the Horus as a student, but not as a trainer/master. So it's not the connection.

matkovits, 2021-10-04:
that's right
horus as a student is ok
Horus as a teacher does not work

3djc, 2021-10-04:
> @3djc : No ... @matkovits can use the Horus as a student, but not as a trainer/master. So it's not the connection.

That’s the opposite. Master listen to signal, and only does so when the detect pin gets activated…

matkovits, 2021-10-05:
Where is the "detect pin" and how i can i activate it

rotorman, 2021-10-05:
In the trainer connector. For code definitions, see here:
https://github.com/opentx/opentx/blob/bfe94fc42f3984e943429785cc4849b75d51f200/radio/src/targets/horus/hal.h#L884-L885

Target0815, 2021-10-05:
@rotorman, @matkovits does not mean the pin on the CPU, but the "detect pin" on the socket that @3djc mentioned. The latter is also not clear to me, because normally when plugging in either a contact is opened (!) or switched. There is no extra pin. Maybe @3djc can explain this even better ...?

3djc, 2021-10-05:
You can have a look here : https://github.com/opentx/opentx/issues/6245#issuecomment-440562543