opentx/opentx #8761: No RF Freq Fine tune option after updating to 2.3.14

RavenOneEightSeven, 2021-10-30:
I recently flashed from the original Jumper t16 version of open TX to 2.3.14

Everything works fine rx wise, but I noticed today that there is no freq fine tune option.

**RX = FRSKY xm+**

I**nternal multi module - version 1.2.1.51 (the older version, without the bootloader)**

**I am getting the 'Module update recommended' prompt because it's the old version.** 

Quad is bound, failsafe works and I have been flying all my old models. I started noticing some weird failsafes and decided to check things. I can really only pinpoint the lack of freq fine tune for this (as I also had strange failsafes before I freq fine tuned on the old version of openTX).

3djc, 2021-10-30:
Options are now sent by the module, that asks OTX to display or not options depending on protocol. You must update the module for normal operation

closed 2021-10-30 by 3djc

RavenOneEightSeven, 2021-10-30:
Thanks for the fast response and clarification. Any idea when this was changed? I might just roll back, as I would need to get an FTDI adapter to first install the bootloader etc etc.