opentx/opentx #8764: Trainer Socket Detection on QX7

Henryford70, 2021-11-02:
Just bought two QX7's (running on V2 firmware - 2.3.14) as I'm learning to fly fixed wing, one set-up as "Student" (slave) and the other as "Instructor" (Master). When 3.5mm audio lead plugged in master TX doesn't detect the slave TX. On two occasions  it did detect it momentary but it wasn't plugged in fully. Thinking I have bought a defective TX, I bought another QX7, same symptoms as above.
A fellow flyer who has a QX7 running on V1 firmware, detected my student TX straight away.
Is there an issue with firmware detecting trainer socket on QX7? I notice 2.3.14 was released to cure a similar issue with the X9D TX.

justtrashmaker, 2021-11-10:
Hallo,

I have the same problem. I tested an older version (2.3.9). It works well here.

ChrisOhara57, 2021-11-10:
Hello
it seems to be the same problem I had.  #8670
In my case it is a x9lite S.
I worked on 2.3.7 but did not on 2.3.14
I flashed 2.3.11 and it does work now.

3djc, 2021-11-10:
Dupliacte #8670

closed 2021-11-10 by 3djc