opentx/opentx #8765: 2.3.14 (1e09791a) Instant Trim Doesn't Work for Existing Models

jmxp69, 2021-11-03:
2.3.14 (1e09791a)

Radiomaster TX16s
Test 1: Instant Trim Defined in Global Functions via momentary switch. Bold GF line confirmed on activation of switch.
Test 2: Instant Trim Defined in Special Functions via momentary switch. Bold SF line confirmed on activation of switch.

I've been using instant trim for 5+ years in global functions activated by a momentary switch.
Using an existing model with trims enabled in FM0 and sticks in non-0 positions, instant trim does not apply stick positions to trims using Test 1 and Test 2.

Create new model:
Stocks in non-0 positions, instant trim applies delta from 0 to trim using Test 1 and Test 2.

Trims T1-T6 otherwise work.

Simple summary: Using multiple existing models, instant trim doesn't work. Create new model and no other changes, instant trim works normally.

kilrah, 2021-11-04:
If this can only be reproduced in specific models please include an otx file containing such models.

jmxp69, 2021-11-04:
It's reproduceable on all of my existing models.

kilrah, 2021-11-04:
Still provide your OTX since it doesn't seem to be reproducible with models that aren't yours...

jmxp69, 2021-11-04:
[Instant_Trim_OTX.zip](https://github.com/opentx/opentx/files/7478239/Instant_Trim_OTX.zip)
Model83 is the test model that works. The other two are examples of models I've had on this radio for a while that do not work.

kilrah, 2021-11-05:
For some reason in those 2 models in the inputs you'ce changed the default trim assignment from "Yes" (default for that stick) to manually choosing the corresponding trim. That prevents Instant trim from working.

closed 2021-11-05 by kilrah

jmxp69, 2021-11-05:
Except I didn't make that change. I've never changed that feature on any of my models. All of my pre-existing (65 of them) have the same option.

bigtwin, 2021-11-05:
I guess you are human like all of us. As such you cannot be certain that you never have changed that setting. You just can be certain, that you never wanted to change it.

jmxp69, 2021-11-05:
I'm certain I didn't change that setting for 65 planes.

kilrah, 2021-11-05:
Are those models created via "new model"? Many people just copy an existing one, so it's easy to have one weird thing you did in the original one everywhere.

Only other explanation would be a conversion issue at some point but since you say you've been using these for 5 years it'd be hard to pinpoint when it happened.