opentx/opentx #8770: SD card not found on Windows 10 (Virtual Machine)

loman1978, 2021-11-06:
**Describe the bug**
I want to connect the Radio to my Windows 10. Windows 10 is a virtual machine in my MacOS. I'm using Parallel Desktop to virtualize Windows 10. When I try to read something from my radio I get the message "radio SD card not found". I suppose the program (last version) is not able to find the SD card mapped in the virtual machine because are not a real storage drivers but are shared folder: is it possible? In the virtual machine I can see 2 folders: TX16 (with 2 files) and the SD card content.
When I use MacOS the companion application works correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the boot mode in the radio and connect the USB cable
2. Wait the 2 USB driver "NO NAME" and "TX16S" on Mac OS
3. Run the virtual machine and be sure the 2 drivers are shared 
4. Try to read the radio content: you should see "Unable to find radio SD Card"

**Expected behavior**
I expect the program is able to identify the SD card even if is a shared folder.

**Screenshots**
Here the error I get when I try to read the radio content
[image]

Here the 2 driver I get from the radio
[image]

Here the content of the first storage driver of the radio (as shared folder)
[image]

Here the content of SD card of the radio (as shared folder)
[image]

**General information**
 - OpenTX version: 2.3.14
 - Hardware: RadioMaster TX16S
 - OpenTX Companion: 2.3.14 on Windows 10)
- MAC: Mac book Pro with Windows 10 on Parallel Desktop
-

3djc, 2021-11-06:
Why are. It using the Mac version ???

bigtwin, 2021-11-06:
I did it once using a virtualized _Windows10_ on MAC OSX. But it was with _VMware Fusion_. Everything was working as expected. Which version of MAC OSX are you using?
Have you checked _Parallels_ config? May the SD card is passed read-only to the VM-Guest. Or something similar. Have you already done some tests? With a different Transmitter or SD-Card? Has ist ever worked?
I had trouble with bad USB-Cables. I also had reproducable, different behavior with Virtualizations when using USB2 instead of USB3 and vice versa.

@3djc - in my situation for example, it was this [issue](https://github.com/opentx/opentx/issues/8561) which made me trying the Windows version in a virtualization on a MAC.

kilrah, 2021-11-06:
If you use a VM you'll want to pass through the USB device, that's the way for the VM to see things as if it was running native.

loman1978, 2021-11-07:
I'm using the latest version of MacOS (big sur) and I think is not a problem related to the cable or SD card. In the virtual Machine I can see the drivers and look at the content... Probably I need to change how to "pass" the USB to the virtual machine... probably I should not share the folders but let windows to manage the USB completely 🤔🤔🤔 I don't know how...

bigtwin, 2021-11-07:
In the configuration of _Parallels_ there must be somewhere a point where you can choose who should see the physical device - the host or the guest (configuration/options of the VM). Change this to (Windows) guest only.  Somewhere in that corner, there is probably also the point to configre whether your Windows Guest should see a USB2 or a USB3 device. Change this vice versa for testing.
Also, I highly recommend to test a well known working transmitter. Has it ever worked with your RM?
As you can see the SD-Card in Windows - would you be able to write something to SD-Card usiing the Windows? Just try to copy a test file to it. 
If you install Companion on your Big Sur - would that properly work?

The latest version of MAC OSX is Monterey, not Big Sur. I am using Mojave (with VMware Fusion).

bigtwin, 2021-11-07:
I just did a quick test ...
Companion 2.3.14 works fine in my Win10 _VMware Fusion_ Guest Operating system on my 2015 MacBook Pro with _MacOSX Mojave_ installed. I could managed my QX7 just like in Companion native on OSX.

This points again towards an issue with your local setup or the involved components. You definetly need to do some more testing in order to locate the culprit.
Though, it still can be an issue with Companion on Big Sur, but I would not assume such an error being related to that.

loman1978, 2021-11-07:
@bigtwin Yes sorry not Big Sur.. I'm using Monterey... I was confused 😁. It works in the Mac but not in the virtual Machine

loman1978, 2021-11-07:
Solved!!! I was looking in the wrong place... When I run the Virtual machine, there is the USB device called "Radiomaster TX16S bootloader". I can assign this device to the virtual machine but then I need to change the driver in Windows! This is what I did:
- Opened the device in the control panel
- Selected "Update driver"
- Select manually the driver
- Windows propose different driver.. I need to select Mass Storage driver
- Apply.

After , windows 10 automatically mounted the 2 mass storage device: "NO NAME" and "TX16S" . The companion App now works correctly.

Thanks to all for the suggestions
I will close this issue

closed 2021-11-07 by loman1978