opentx/opentx #8771: Support for new position telemetry sensor data in Archer SRx Receivers

rieden1, 2021-11-08:
[apologies if I am doing this incorrectly - I am new to this process. I did try searching using a selection of terms but didn't find anything]

FrSky Archer receiver firmware 2.1.9 added new sensor telemetry data for 3-axis acceleration and 3-axis angular position from Stabilised receivers. When these sensors are discovered in OTx (2.3.14, Horus X10sSE) the three acceleration sensors are correctly identified as AccX/Y/Z, but the three angular sensors are not. Instead there is a fourth sensor called "0730" which displays an 8-digit signed integer value - I'm guessing it is comprised of all three values perhaps coded into a 24bit binary number for transmission or something):

[image]

This is a request for support for this data in OTx. If it would help I would be more than happy to provide any test/experiment/investigation to support if needed.

raphaelcoeffic, 2021-11-08:
A specification maybe as well? Or is reverse engineering required?

rieden1, 2021-11-08:
I don't have the specification I'm afraid. I don't know if FrSky will supply it, so it might need reverse engineering.

pulquero, 2021-12-08:
AccX/Y/Z are three individual sensors (0700,0710,0720), not 3 values from a single sensor. (Smartport) sensor values are 32bit (little endian). Values are normally integers with a predetermined fixed point. One interesting sensor is GPS - the lat/long are sent alternately with the msb set to 1 as a flag for longitude.
Hopefully this will help your decoding efforts.

rieden1, 2021-12-08:
Apologies - I have obviously not made myself clear. The three acceleration sensors (AccX/Y/Z) are "discovered" and function normally. It's the three angular position sensors that seem to be combined into the single sensor "730". That's the value I'm trying to decode.