opentx/opentx #8773: Telemetry simulator unable to simulate voltages for Omphobby + Radiomaster TX16s

LevYas, 2021-11-09:
**Describe the bug**
I discovered the telemetry sensors while connected to the model, tried to simulate voltages, it didn't work, values on the radio in the simulation are grayed out and all zeros.
Also, tried to write some logs and replay them, the same effect - no voltages in radio simulation

**To Reproduce**
Steps to reproduce the behavior:
1. Get OMP M2 and Radiomaster TX16s and set them up using the OMPHOBBY protocol
2. Discover the model's sensor using the actual radio (not simulation)
3. Connect the radio to the Companion
4. Read radio's params
5. Start radio simulation
6. Open telemetry simulation
7. See that RSSI level changing works, but A1 and A2 don't - no matter what you put in the fields for A1 and A2, the values on the radio are all 0s. 

**Expected behavior**
I expect to be able to simulate voltages to verify how I set up voltage alerts and flight counters.

**Screenshots**
[image]

**General information**
 - OpenTX version: 2.3.14-otx 2021-07-23
 - Hardware: helicopter OMP M2 and Radiomaster TX16s
 - OpenTX Companion Version: 2.3.14, Oct  1 2021

ChrisOhara57, 2021-12-04:
It's not clear to me what you mean.
You write "no matter what you put in the fields for A1 and A2".
Have you written the value 12.8 V manually into the field or are they part of the logfile?
It looks as if you are replaying a log-file so the values shown hear should be the ones of the log-file.
Are you sure that you have clicked the checkbox "logs" of the A2 value (prior to the flight) so that is part of the log-file.
(at the right end of the line)

[image]

kilrah, 2021-12-04:
Seems this aircraft sends telemetry with differnt IDs than FrSky, so that's what the sensors get set to. The player / simulator however puts out the FrSky IDs so they don't get seen.

ChrisOhara57, 2021-12-04:
If that's the reason you could try:

**in Companion**
Make a copy of your model.
In the copied model, change the sensor ID of A2 to the standard Frsky ID. (see screenshot of my first message)
Now run the simulator with your logfile.

LevYas, 2021-12-04:
> It's not clear to me what you mean. You write "no matter what you put in the fields for A1 and A2". Have you written the value 12.8 V manually into the field or are they part of the logfile?
On the screenshot, they're from the log file. I tried to enter manually something, no luck. All the telemetry goes to the logs. I can send the logs if needed.

> The player / simulator however puts out the FrSky IDs so they don't get seen.
oh, I screwed. Looked at the code https://github1s.com/opentx/opentx/blob/HEAD/companion/src/simulation/telemetrysimu.cpp#L224 
indeed, all definitions are from frsky.h

> change the sensor ID of A2 to the standard Frsky ID 
Hm, that should work, so I will have one "main" model and one to practice telemetry-related scripts :) though much better than nothing, thank you.

LevYas, 2022-06-13:
Flexible telemetry simulation would still be great