opentx/opentx #8778: Enable CRSF protocol on internal sport

sdellava, 2021-11-15:
This issue came from the difficulty of configuring the ELRS protocol on a 5in1 internal module jumper. After a few attempts by the ELRS devs it became clear that this is not possible, either due to software problems or hardware compatibility.

Since ELRS is suitable to replace many protocols reusing the receivers, I decided to change the approach trying to replace the internal module with a native ELRS one. Why not use an external add-on module? Because being able to replace the largely useless internal one, the external bay remains free to install a JR module when needed.

The only practical problem to complete this project is that the ELRS modules communicate using the CRSF protocol, already supported in OpenTx but enabled only for external modules.

A team of hardware designers is already working to realize an internal module for the Jumper T18, with 868/915 MHz and 2.4 GHz band (dual band!), compatible with the connection and size.

The request is therefore, simply to enable the CRSF protocol on the configuration menu of the receiver to install and use this module.

raphaelcoeffic, 2021-11-16:
@sdellava see here: https://github.com/EdgeTX/edgetx/pull/1039. I hope this is coming as well for OpenTx!

sdellava, 2021-11-16:
> @sdellava see here: [EdgeTX/edgetx#1039](https://github.com/EdgeTX/edgetx/pull/1039). I hope this is coming as well for OpenTx!

Good. Will this update be valid for every radio or only for Tango ?

raphaelcoeffic, 2021-11-21:
> > @sdellava see here: [EdgeTX/edgetx#1039](https://github.com/EdgeTX/edgetx/pull/1039). I hope this is coming as well for OpenTx!
> 
> Good. Will this update be valid for every radio or only for Tango ?

It’s already working on the TX16S, T16, and probably all derivates. It should work the same on all targets with a duplex serial port on the internal interface.