opentx/opentx #8780: Jumper T-Lite switch locations

FelixSchreiber, 2021-11-19:
In the radios Main View the switch locations are not ordered according to the real hardware switch locations.

On the radio, C and A are left to the antenna,  B and D are right to it; C is the leftmost and D the rightmost switch. C and D are located more to the back whereas A and B are located more to the front of the radio.
In the main view, A and B are left to the left stick, C and D are right to the right stick.
[image]
A more intuitive order would e.g be :
Left to the left stick: First Switch A and below switch C
Right to the right stick: First Switch B and below switch D

In Companion the switches are ordered A, B, C, D (from left to right)
[image]
A more intuitive order would be according to the real switch locations: C, A, B, D (from left to right)

Furthermore it would be very helpful if the other buttons in the radio image could be named (as long as you are not fully used to this radio and know it by heart).
[image]

**General information**
 - OpenTX version 2.3.14
 - Hardware Jumper T-Lite
 - OpenTX Companion .3.14

3djc, 2021-11-19:
Switch ate alphabeticaly ordered by design

closed 2021-11-19 by 3djc

FelixSchreiber, 2021-11-19:
sorry, but this answer is too simple and doesn't comply with reality (probably you are referring only to the Companion?).

In radio/src/gui/128x64/view_main.cpp there are for example many different arrangements for the switches defined.
Why not add something similar for this radio like

```
#elif defined(PCBXLITES)
        static const uint8_t x[NUM_SWITCHES] = {2*FW-2, 16*FW+1, 2*FW-2, 16*FW+1};
        static const uint8_t y[NUM_SWITCHES] = {4*FH+1, 4*FH+1, 6*FH+1, 6*FH+1};
        for (int i=0; i<NUM_SWITCHES; ++i) {
          if (SWITCH_EXISTS(i)) {
            getvalue_t val = getValue(MIXSRC_FIRST_SWITCH + i);
            getvalue_t sw = ((val < 0) ? 3 * i + 1 : ((val == 0) ? 3 * i + 2 : 3 * i + 3));
            drawSwitch(x[i], y[i], sw, 0, false);
          }
        }
```

FelixSchreiber, 2021-11-19:
this solves at least my first point

[image]

```
diff --git a/radio/src/gui/128x64/view_main.cpp b/radio/src/gui/128x64/view_main.cpp
index b3cea8de7..cb3bc05fc 100644
--- a/radio/src/gui/128x64/view_main.cpp
+++ b/radio/src/gui/128x64/view_main.cpp
@@ -534,6 +534,16 @@ void menuMainView(event_t event)
             drawSwitch(x, y, sw, 0, false);
           }
         }
+#elif defined(RADIO_TLITE)
+        static const uint8_t x[NUM_SWITCHES] = {2 * FW - 2, 16 * FW + 1, 2 * FW - 2, 16 * FW + 1};
+        static const uint8_t y[NUM_SWITCHES] = {4 * FH + 1, 4 * FH + 1, 5 * FH + 1, 5 * FH + 1};
+        for (int i = 0; i < NUM_SWITCHES; ++i) {
+          if (SWITCH_EXISTS(i)) {
+            getvalue_t val = getValue(MIXSRC_FIRST_SWITCH + i);
+            getvalue_t sw = ((val < 0) ? 3 * i + 1 : ((val == 0) ? 3 * i + 2 : 3 * i + 3));
+            drawSwitch(x[i], y[i], sw, 0, false);
+          }
+        }
 #else
         // The ID0 3-POS switch is merged with the TRN switch
         for (uint8_t i=SWSRC_THR; i<=SWSRC_TRN; i++) {

```

PR #8782 (2021-12-09): Improve view main
This fixes #8780