opentx/opentx #8786: X9D+ SE 2019 and multimodule communication cut off with SF switch when openTX >= 2.3.12

EddyBag42, 2021-11-30:
Hi, 
- communication between radio and module is set in serial mode (MULTI)

I have a Taranis X9D+ SE 2019 (opentTX 2.3.14) and an iRangeX IRX4+ multimodule (firmware 1.3.6)
I'm facing a fairly critical issue of communication cut off between radio and module.

In serial mode (MULTI), communication between the radio and the multimodule cuts off under reproducible circumstances as follows :
- radio is using external RF 
- communication between radio and module is set in serial mode (MULTI)
- receiver doesn't need to be on, though problem may appear quicker when receiver is on (not sure of this point)
- after several toggles of the 2-positions switch SF (random number of toggles, that can be one or ten, rarely more than a few tens) the communication between the radio and the multimodule cuts off (the radio says "No serial input" and the red led flashes slowly on the multimodule)

[image]

I posted this issue to RC Groups thread and it appears that this is not an isolated case, it seems to be related to using multimodule inside X9D+ SE 2019, please [see here](https://www.rcgroups.com/forums/showpost.php?p=48240461&postcount=23142).

[Old openTX versions](https://www.rcgroups.com/forums/showpost.php?p=48249615&postcount=23150) for X9D+ SE 2019 have been posted on the same thread so I could try and see if issue is still there.

Here are the results of my tests : 
- openTX 2.3.13 : issue is present
- openTX 2.3.12 : issue is present
- openTX 2.3.11 : no issue detected
- openTX 2.3.10 : no issue detected
- openTX < 2.3.10 : not tested

So it seems to me that the issue appeared with openTX 2.3.12 and is there since.

Thanks for your help. Best regards.

Remark : no issue with another switch than SF, no issue with PPM communication mode, no issue with SF switch when using internal RF, no issue with an X9D, seems specific to X9D+ SE 2019 + multimodule + serial mode + SF switch

EddyBag42, 2022-06-13:
Still present bug with 2.3.15, stuck with 2.3.11

gratefulfrog, 2022-08-18:
I confirm that this bug is there and terrible

gratefulfrog, 2022-09-13:
any progress??

Thanks,
B

HippoDan, 2023-07-22:
This is still an ongoing problem.  Lost a plane this week because of it.  Going back to 2.3.11 for the time being.  Please fix this, I don't want to be stuck on older firmware forever.

bsongis, 2023-09-21:
Do you use SF for anything? Like screenshot

gratefulfrog, 2023-09-21:
Yes, I use SF for throttle cut on all my models.

Thanks for your help, it would be great to fix this so we can move to the
latest version of open-tx!

bsongis, 2023-09-21:
Do you reproduce if you use SF for nothing?
And on a blank model?

gratefulfrog, 2023-09-21:
Yes, it still happens on a blank model with nothing assigned to SF.

HippoDan, 2023-09-21:
> Do you use SF for anything? Like screenshot

I like my momentary switch on the left, and I use it to call out my timer among other things.  It gets pressed a lot for timer, gun sound effects, self level and more.  Started having the problem right away, but it took forever for me to figure out what was happening.

gratefulfrog, 2023-09-21:
So are you able to work on a bug fix? This problem has been open for a
while and no one seems to be looking at it...

I would look at the code, but it would take me so long to dive in... I
hardly have the time.

Let me know about the EEPROM if you could use it - it is version 219 I think

Ceeb182, 2023-09-21:
The EEPROM is the complete copy of the radio's contents (models + radio configuration).

To obtain the EEPROM and attach it here in a message: 
1. Install Companion 2.3 software on a PC
2. Put the radio in "Bootloader" mode (press the two horizontal trims towards the center while powering up the radio, the screen will turn on with "OpenTx Bootloader" in the title bar).
3. Open the Companion software on PC (if the software suggests updates, ignore them for the moment, i.e. answer "No" twice).
4. Connect the radio via USB to the PC. The radio displays "USB Connected".
5. In Companion software, go to menu: Read Write > Backup Radio To File
6. Choose an explicit file name (Example: "OTX2313-BugSF-ModelName.bin")
7. Companion displays a bar graph showing the progress of the EEPROM transfer to the file "OTX2313-BugSF-ModelName.bin".
8. Attach the file to this issue's message (by dragging and dropping "OTX2313-BugSF-ModelName.bin" in the message)
9. Specify in the message: radio type, radio's OpenTx version
10. And that's it!

bsongis, 2023-09-21:
Someone reproduces the issue with a radio with the small external module bay? because I only have the small Multimodule ...

gratefulfrog, 2023-09-21:
As requested, I attach my EEPROM File, but please note that this is OTX
version 2.3.11 which is the most recent version that DOES NOT HAVE THE BUG.

The bug is present in all versions 2.3.XX where XX > 11.

I do not use the more recent version because of the bug!

My firmware is:
FW: opentx-x9d+2019
VERS: 2.3.11-OTX (4328055e)
Date: 2021-01-08 23:44:58
EEPR: 219

I hope this helps, I am so glad that you guys are working on it. Let me
know if I can help, I am a good programmer and debugger...
Cheers,
Bob

bsongis, 2023-09-21:
Searching for a module to reproduce the problem ...

bsongis, 2023-09-22:
Now I have X9D+ updated to latest 'master', compiled with a recent compiler (the one I use for Ethos)
I have a multimodulle in version 1.3.3.20 AETR
I cannot reproduce the problem

bsongis, 2023-09-22:
Now the same opentx, compiled with the old good compiler. I cannot reproduce the problem.

bsongis, 2023-09-22:
Now the official firmware downloaded from companion. I cannot reproduce the problem.
I have not bound the RX, should I?
Which options did you have?

bsongis, 2023-09-22:
@gratefulfrog can you also share your eeprom here?

bsongis, 2023-09-22:
Now with the official firmware downloaded from companion and bound with a Spektrum AR410.
I cannot reproduce the problem

EddyBag42, 2023-09-22:
On Fri, Sep 22, 2023, 06:44 Bertrand Songis ***@***.***>
wrote:

> Now I have X9D+ updated to latest 'master'
>
Hi,
Is it a X9D+ SE 2019 ?
You will not have thé problem with a regular X9D+.
Cheers.
-- 
Ed

bsongis, 2023-09-22:
Yes of course

bsongis, 2023-09-22:
[image]

bsongis, 2023-09-22:
No luck, I have the multimodule marked as "No problem"
I also have the iRX4 lite, searching for a way to connect it to the fullsize module slot...

3djc, 2023-09-22:
> On Fri, Sep 22, 2023, 06:44 Bertrand Songis ***@***.***> wrote: Now I have X9D+ updated to latest 'master'
> Hi, Is it a X9D+ SE 2019 ? You will not have thé problem with a regular X9D+. Cheers. -- Ed

At this point in time, we cannot reproduce and we spend time going through all the differences between 2.3.11 and 2.3.12 and found nothing that could explain that

3djc, 2023-09-22:
> On Fri, Sep 22, 2023, 06:44 Bertrand Songis ***@***.***> wrote: Now I have X9D+ updated to latest 'master'
> Hi, Is it a X9D+ SE 2019 ? You will not have thé problem with a regular X9D+. Cheers. -- Ed

If you have the issue too, an eeprom back would be helpful

bsongis, 2023-09-22:
Now testing with iRX4, bound to the AR410
Still no issue

3djc, 2023-09-22:
At this point in time, we cannot find any issue. Unless we get a non working eeprom, we will stop investigating

gratefulfrog, 2023-09-22:
Hello,
I am sorry that you have not been able to reproduce the bug.

It simply requires a Taranis X9D+ 2019 SE and any version of OTX 2.3.12 or
more recent and an IRX4 multimodule with latest firmware.

I  could try to flash the new OTX but I am very worried that my models will
be ruined or lost when I do that.

Is there a procedure that would ensure that I could go back to the exact
same configuration that I now have? And are you sure that it will work?

I would be destroyed if I had to reprogram my radio and all my models...

Thanks for your help:
Cheers,
Bob

EddyBag42, 2023-09-22:
gretefulfrog is perfectly right

I can't figure out why you can't reproduce this issue, it's incredibly
frustrating and too bad

I'll try to upgrade to latest firmware and see whether the issue is still
present or not. If I get the issue (which I think I will for now) I'll post
the eeprom as requested. I'll do that as soon as I can, in the next hours
or so.

Thanks

EddyBag42, 2023-09-22:
As I've already said earlier in this thread, I confirm the SF bug is still there with openTX 2.3.15 and this particular radio and module.

[Here you have a zip](https://github.com/opentx/opentx/files/12703513/OTX2315-BugSF.zip) including 3 files :
- a picture of my radio with problem shown ("No serial input") so you can make sure the radio you're using is the same PLUS SE 2019 (carbon or blue, hall effect sticks and ACCESS)
- a settings and models backup made with "Backup radio to file"
- a firmware backup made with "Read firmware from radio"

You can use Model 15 named "test sf" to test

Hardware : X9D PLUS SE 2019 and iRangeX IRX4+ multimodule
Software : opentTX 2.3.15

Thanks a lot for your investigations.
Hope it will help.

bigtwin, 2023-09-22:
Maybe this is not related to the OTX-Version. May it is something different? An issue with hard- or firmware (none OTX)? Something which just and only strikes in >2.3.12 not because of an error in OTX, but some other option/feature/timing ...

gratefulfrog, 2023-09-22:
What could explain that the problem only occurs from otx 2.3.12 onwards
other than a change in otx? There is nothing else that changed, is there? I
cannot see any other explanation.

In the 2.3.12 release notes I see a few points which are suspicious:
- add SI support to X9D 2019
- Always prioritize protocol data coming directly from the module
- Add support for module config (to allow two multimodule to use the same
ID for example)
- Update data tables to 1.3.2.84 (thanks MRC3742)

Perhaps there is some strange conflict created by these and/or other
changes?

3djc, 2023-09-23:
If it was a clear cut error, it would affect all multi and would be easy to reproduce. It is neither

bigtwin, 2023-09-23:
This issue seems to be extremly specific ...

Switch SF ... may it has nothing to to do with SF but the 6th switch or the 4th being used, or the one before SG, or the switch with a cabling that is close to some point at the main board. Or switch chattering is not prevented by an capasitor.

Exact this type of Transmitter? May all your guys trasmitters were manufactured the same time? Equipped with the same cabling/switch or assembled by the same person (critical cable placement)?

Since it is that specific, I would think this is not direct software related, maybe indirect. The code has changed since 2.3.12 and now some other critical condition strikes in and causes this malfunction. It is obviously sophisticated and "since 2.3.12" may leads to a wrong approach. If there is an interest to debug such a tricky malfunction, the troubleshooting certainly need to stay open minded.
I would even consider that this problem is not resolvable with available ressources and it may should be covered by documentation/PreReqs.

EddyBag42, 2023-09-23:
It definitely has something to do with SF. Problem does not appear with
other switches. I've done enough testing, and gratefulfrog, and the guy
from RCGroups. It is a fact.

Another fact is that problem appeared with openTX 2.3.12, and is there
since.

It is related to specific hardware that's right. But it's really hard to
believe for me that only several Plus SE 2019 are affected !

Anyway a change in openTX revealed this issue. Maybe if we knew what change
in openTX code caused the issue this would help to understand. What other
way is there ?

This issue is critical. Your risk loosing your model if you touch SF
switch, even the first time you touch it ! It's not only a question of
documentation.

bigtwin, 2023-09-23:
Yes, you are right. But having in mind what limited ressources are available for development and troubleshooting in OTX, it may is just impossible to identify the root cause. Yes, with unlimited ressources it would certainly be possible to resolve this problem technically. But those are not available, therefor documentation is surley a potential solution. May even the only one we can get.

You cannot blame the loss of a model to this bug. Especially not, since it is now a knwon issue. Have in mind that you are using a transmitter and a certain configuration (SF) eventhough you know about this bug. You may should simply stay away from SF or that kind of transmitter.

Dont get me wrong. I agree in all what you wrote, but as said - if it is not possible to identify the root cause, it is probably time to live with it and work around it. This is a common way you cannot ignore. Even commercial manufactures choose this sometimes. Take the Profi TX mishap from Multiplex as an exsample.  
I lost once a model due to some bug in _Spectrum_. Even 100% reproducable. It was committed and confirmed by the dealer and Spectrum, but never clarified to the end. I simply had to stay away from some types of receivers eventhough they were officially documented as being OK.

3djc, 2023-09-23:
We have reviewed with 2 pair of eyes EVERY single line of code different between 2.3.11 and 2.3.12, and found nothing that can explain that.

We also haven't been able to reproduce so far (yes, using very same radio and module). Another attempt will be done Monday, but no luck so far.

EddyBag42, 2023-09-23:
Sure, investigating or not is openTX team's choice, I didn't say anything
différent.

3djc, 2023-09-23:
We haven't give hope yet, but we already spent a lot of time on this issue without been able to find anything

gratefulfrog, 2023-09-23:
I appreciate all that has been said and understand that resources are
limited.

What could explain the problem, if it's not the code? Some bad radios
coincidentally purchased? I myself an a professional in IT and have leadsny
projects. My experience is that the problem is always the code.

Is there anything I could do to help?

Did the compiler options change from 2.3.11 to 2.3.12 by any chance?

3djc, 2023-09-23:
I don't understand that comment on ressource, it doesn't apply here at all.

We reviewed code changes line by line: nothing suspect.

We try to reproduce: impossible so far.

So at this point in time, until we find a way to reproduce (if possible at all), we have no way forward

gratefulfrog, 2023-09-23:
I'm sorry, I meant no offense.

And you cannot reproduce the problem with an frsky x9d+ 2019SE?

I don't know how to proceed either, here we reproduce the problem 100% of
the time.

What can we do to help?

kilrah, 2023-09-24:
> Hardware : X9D PLUS SE 2019 and iRangeX IRX4+ multimodule Software : openTX 2.3.15

What multi firmware? Has it been confirmed across multiple multi firmware versions?
Do the multis exhibiting the problem have something in common e.g. the smaller and/or clone MCU?

bsongis, 2023-09-25:
My multimodule is the iRX4 Lite, flashed with Multi V1.3.2.61AETR
I have flashed the firmware from your ZIP
I have transfered the eeprom from your ZIP
I have bound my RX as you also use DSM
I have toggled SF around 100 times, at various speeds.
Everything works perfectly on my radio, the link is rock solid

gratefulfrog, 2023-09-25:
Hi,

Thanks for allthis work!

Which zip EEPROM did you use exactly? Mine (gratefulfrog) is working since
it is otx 2.3.11, the zip from EddyBag42 is the one with the bug.

I cannot understand how it could work when I did it, after 3 or 4 cliccks
on SF, the comm is lost?

3djc, 2023-09-25:
We tried over 100 SF toggle, all works perfect, despite using the supplied firmware, eeprom, everything

bsongis, 2023-09-25:
@gratefulfrog only EddyBag42 attached an EEPROM here. I installed it with the firmware provided in the same ZIP.
I cannot reproduce the problem. Now I doubt it's only a firmware issue.

gratefulfrog, 2023-09-25:
> I have pushed the problem to FrSky hardware designer, I hope he will have
> an idea

I really appreciate all your efforts!

I just can't find an explanation of how it could fail for both me and Eddy
and a third guy, yet for no one else...

   - our radios are all defective?  would seem unlikely
   - our multimodules defective, again unlikely,
   - maybe your radio is a more recent version of the FrSky X9D+ 2019 SE
   and the hardware is different? But the 2019 SE is only a few years old,
   could FrSky have already changed the hardware spec?

As I said, my experience with decades of software development is that the
problem is always in the code, but this time I am at a loss as to how to
explain that you guys don't have the issue...

I wish I could help....

Cheers,
Bob

HippoDan, 2023-09-25:
Mine is also an IRX4 module on 1.3.3.20.  I can borrow a friend's Jumper module on Wednesday and see if it has the same problem.  I'll upload an Eeprom then.

gratefulfrog, 2023-09-25:
I'm wondering if this whole problem could be a result of the SF switch
bouncing more or less on certain radios? Maybe FrSky changed switches, or
modified the filtering circuit at some point and the bounce is causing
problems for the multi-module ?

Just a thought, based on decades of hardware hacking ;-)

Cheers,
B

HippoDan, 2023-09-26:
I'm not using the stock switch, I replaced it with a momentary pushbutton.  

> I'm wondering if this whole problem could be a result of the SF switch bouncing more or less on certain radios? Maybe FrSky changed switches, or modified the filtering circuit at some point and the [...]

gratefulfrog, 2023-09-26:
Wouldn't that possibly explain why the bug can't seem to be reproduced?

HippoDan, 2023-09-26:
Possibly.  I saw it as more of a reason to eliminate the switch as mine is a different make and it still happens.  It still doesn't explain why it happens only in software 2.3.12 and up.

HippoDan, 2023-09-27:
I'll try to test my friend's Jumper 4-in-1 module with my 2.3.15 Taranis 2019 SE today.   Also, has anyone done this mod to their module?  I'm considering trying that next.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621

HippoDan, 2023-09-28:
I borrowed the Jumper module yesterday.  I clicked over 100 times without the problem.  Turn off/on the radio, clicked 120 times, no problem.  I put the IRX4 back in and clicked 100 times no problem, off/on, then 120 times no problem.  Very confused.  Repeated and the problem reappeared at 50 clicks, then 21 clicks.

Now I don't know whether the Jumper module was a lucky break, or not having the problem.  Intermittent problems are the worst.  How will I ever know if I can trust the radio again?

Anyway, when I see him again I'll test the Jumper module again.  Anybody else have access to a different module?

EddyBag42, 2023-09-28:
Jumper module has been tested with no problem, see [here](https://www.rcgroups.com/forums/showpost.php?p=48240461&postcount=23142).

HippoDan, 2023-10-10:
Strangely it seems the protocol has an effect.  On FrSky V mode it takes anywhere from 1 to 25 presses to cause the problem with it rarely going over that.  On the v761 protocol it usually takes 25-60 presses, with rare outliers for the problem to happen.  I guess the complexity/quantity of data is affecting the outcome.  Some kind of buffer overflow?  Hardware is not my area.

3djc, 2023-10-10:
The dialog between mpm module and radio does not change with protocol, and also is exactly the same for all brands of mpm

gratefulfrog, 2023-10-16:
Hello Guys,

Today I decided to take a risk and update my Frskly Taranis X9D plus 2019
SE to the latest firmware and see if I could reproduce the bug.

The short version is :

   - Yes, the Bug is present : after 46 clicks of the SF switch, the module
   status becomes "no serial input" and the red led starts flashing.
   - Yes, I confirmed this by restarting the radio and again after 59
   clicks, "no serial input".
   - I attach the EEPROM file which shows the bug in the model MultTest
   with the IRangeX Multimodule

Here is the exact process that I followed:

   1. using companion, I downloaded the latest version of  OTX: 2.3.15 and
   configured the options : opentx-x9d+2019-afhds3-flexr9m-lua-en-2.3.15-otx.
   2. I downloaded the content of the corresponding SD card
   3. I  made a backup copy of my current SD card
   4. I made a backup of my models and settings
   5. I made a backup of my radio
   6. I copied the content of the new SD card to the physical card
   7. I added the files that were specific to my old sd card to the new
   one, putting them in the new directories as needed and deleting all the
   complied versions of the lua scripts
   8. I flashed the latest firmware to the radio with the companion.
   9. I restarted the radio, physically installed the IRangeX module and
   selected the MultTest model
   10.  I observed on the model configuration menu

External RF:
Mode : MULTI
Type Frsky D D8
Module Status  CH1-16
Receiver 20 [Bind] [Range]

   1. I observed that the red and green leds were lit and solid.
   2. I began to slowly click the SF switch.
   3. After 46 clicks the display showed:

Module Status  No serial input
the RED LED began to flash

   1. I shut down the radio
   2. I restarted the radio
   3. I observed the correct Module Status
   4. I began clicking the SF switch slowly
   5. After 59 clicks, the status changed to "no serial input" and the RED
   LED began flashing

I  believe that we can be sure that the bug is still present.

I sincerely hope that this helps with the search for the problem. Let me
know if there is anything I can do to help!

Thanks for all the effort that everyone is putting in!!
Cheers,
Bob

3djc, 2023-10-16:
Why do you have afhds3 ? Do you have such module ?

[19 more replies]