opentx/opentx #8787: SD explorer shows junk below file list after deleting file

non-bin, 2021-11-30:
**Describe the bug**
When deleting a file and scrolling past the bottom, a list of files named `WWWWWWWWW` are present.
Only affects file lists longer than the screen. Files are still correctly selected while the bug is active

**To Reproduce**
Steps to reproduce the behaviour:
1. Open the SD Explorer
2. Scroll or wrap to the bottom
3. Scroll to and delete a file at least 2 entries above the top of the screen
4a. Scroll to the bottom. There will be W files equal to `files scrolled past top - 1`
  OR
4b. Scroll to the top. The menu will skip the top files, again equal to `files scrolled past top - 1`
5. To clear the bugged state, select the `[..]` entry or page selector

**Screenshots**
<video src="https://user-images.githubusercontent.com/36313060/144134441-20989af7-a610-4a6f-9162-a6770e57e2f1.mp4" />

**General information**
 - OpenTX version 2.3.14-otx (1e09791a)
 - Hardware Jumper T-Lite
 - OpenTX Companion 2.3.14 (Oct 1 2021)

alm-nl, 2022-01-01:
Also happens on X9D+. Not specific for 2.3.14, happens in earlier versions also.

alm-nl, 2022-07-10:
It wasn't marked as fixed, please keep it open till it's being added to the todo list.