opentx/opentx #8788: Companion Enhancement: On the Outputs, Logical Switches, and Special Functions tabs, make column labels stay in view.

Miami-Mike, 2021-12-04:
On Companion's Outputs, Logical Switches, and Special Functions tabs the column labels are attached to the tops of the columns so that when you scroll down they disappear off the top of the page. They should instead be attached to the top of the page so that they will always be visible.

bigtwin, 2021-12-04:
Good point. I surely missed that quite some times!

elecpower, 2021-12-05:
A fix for that was tried a few years back however it affected the application performance significantly and was never progressed.
The change was to use Qt table rather than Qt grid. One would assume no real difference but there was.

bigtwin, 2021-12-05:
Too bad. An improvement like this is surely not worth a significant drop of performance.