opentx/opentx #8794: FeatureRequest: directly reboot into boot loader

bigtwin, 2021-12-12:
When working with my transmitter itself, I have to use the transmitter with booted OTX but really often I need to switch into the boot loader to connect the transmitter with OTX.
Currentyl I have to do this by turnung off and back on the transmitter.

It would be very nice and convenient, if there is a chance to directly leave OTX and boot into the boot loader. Just like I currently can boot OTX when I exit the boot loader.

kilrah, 2021-12-12:
You don't need to reboot to bootloader to access the SD/settings.

bigtwin, 2021-12-12:
Oh man. I am with OTX since 2,5 years now ... I was not aware of this. Thx for that little hint! 
Small thing, big impact to me.

closed 2021-12-12 by bigtwin