opentx/opentx #8802: mode change from the external module puts the transmitter in emergency mode

TheIsotopes, 2021-12-18:
**Describe the bug**
Use of the current master leads to the following errors, back to release 2.3.14 and everything works as expected.

The transmitter goes into emergency mode while a receiver is flashing.
When changing the mode from ppm to another mode, the transmitter also falls into emergency mode.
Emergency mode can only be exited by removing the battery or flashing the transmitter firmware again.

I found out that this bug has existed since #8665. @tbs-fpv pls can you check this.

**To Reproduce**
Steps to reproduce the behavior:
- Select and activate PPM mode
- now try to select another mode
- Transmitter crashes and goes into emergency mode

**Expected behavior**
switching the mode or flashing a receiver should work without a problem.

**General information**
 - OpenTX version latest master
 - Hardware Radiomaster TX16S
 - OpenTX Companion 2.3.14

3djc, 2021-12-19:
Could you let us have the otx file for your radio ?

TheIsotopes, 2021-12-19:
@3djc here is the version created directly from said commit.

[opentx-tx16s-697184a5.zip](https://github.com/opentx/opentx/files/7740487/opentx-tx16s-697184a5.zip)

3djc, 2021-12-19:
No, I would like your radio and model settings please, So get model and settngs with companion and save as otx

TheIsotopes, 2021-12-19:
sorry, misunderstood. Here is the file.

[opentx_mdl.zip](https://github.com/opentx/opentx/files/7740499/opentx_mdl.zip)

robustini, 2021-12-19:
Yes, same issue here with X10S Express.

closed 2021-12-21 by 3djc

TheIsotopes, 2021-12-21:
works again ... thx @3djc fixing this 👍

commit 685657c8 (2021-12-21): Fix #8802