opentx/opentx #8809: Volume adjustment a moment to late on Horus

bigtwin, 2021-12-24:
**Describe the bug**
I have set a _Global Function_ to adjust the transmitters volume using a potentiometer. After the starting sound "_Welcome to OpenTX_" the volume gets adjusted. 
This volume adjustment strikes in a little late. If e.g. the volume is set to zero (muted) you can still hear the first part of whatever the model configuration desire to say (like e.g. _Engine Off_). 

**To Reproduce**
- Configue some sound for the default model, which gets played right away like e.g. some switch position (e.g. _Engine Off_) 
- using the special function "Volume" as Global Function to adjust the volume and adjust it to zero or at least very quietly 
- Power On the transmitter - you will hear the first syllable of e.g. _Engine Off_

**Expected behavior**
I would expect that function "volume" controls the volume from the very beginning after the startup sound (Welcome to OpenTX). As an additional improvement it may even controls the loudness already from the startup sound.

**General information**
 - OpenTX version 2.3.14
 - Horus 10s
 - OpenTX Companion 2.3.14 on MAC OSX Mojave

3djc, 2021-12-24:
That would mean slowing down the whole boot process. Simple personal fix : insert blank period in  your hello.wav sound

bigtwin, 2021-12-24:
If this is for known reason and judged to be as it is - than its fine with me. This is surely a minor issue and I can live with it.

closed 2021-12-24 by bigtwin