opentx/opentx #8812: TBS MAMBO flash external module not working (elrs always no sync)

Sekilsgs2, 2022-01-06:
Latest opentx installed to tbs mambo.
I'm trying flash elrs bootloader to r9m - getting always data refused error
I'm trying flash elrs firmware to r9m - getting always nosync error 
Probably telemetry driver have bug?

3djc, 2022-01-06:
Isn't tbs still using freedom tx ?

Sekilsgs2, 2022-01-06:
> Isn't tbs still using freedom tx ?

I'm install opentx to mambo and tested it myself.
All working good besides flashing external modules..

Sekilsgs2, 2022-01-06:
Oh.. i'm find that tbs telemetry driver loose this 
**sportWaitTransmissionComplete();**
in
**void telemetryPortSetDirectionInput()**
function - i'm think this reason..

Sekilsgs2, 2022-01-06:
I'm add this to telemetry driver but now have another problem - while writing FW to module i'm getting  random errors and writing breaks at random state (5% or 70% and etc).
I'm try stopping crossfire task and this helps but not 100% (now i'm success flashing 2 time and 5 time have errors while writing)..

Sekilsgs2, 2022-01-07:
Upd - adding **sportWaitTransmissionComplete();** fix this bug all other problems!

3djc, 2022-01-07:
you mean that : https://github.com/opentx/opentx/pull/8813 ?

Sekilsgs2, 2022-01-07:
> you mean that : #8813 ?

Yes thank you!

PR #8813 (2022-01-07): Fix TBS Manbo flashing external module 
This fixes #8812
commit 43f23bcb (2022-01-07): This fixes #8812 (#8813)
closed 2022-01-07 by Sekilsgs2