opentx/opentx #8815: TBS Mambo - no pots calibration menu

Sekilsgs2, 2022-01-08:
**Describe the bug**
TBS Mambo - no pots calibration menu (button not active)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to radio hardware page
2. Scroll down to calibration for pots
3. See that cursor skips calibration and go to S1 pot. 

see video
https://mega.nz/file/xSxElYja#xkdeHZqorKBFcjvxdTdk91OxL8BvUQdhKFLJH0YNzlM

 - OpenTX version (latest)
 - Hardware (tbs mambo)

kilrah, 2022-01-08:
Sticks and pots calibrate at the same place, issue is likely the stray label

3djc, 2022-01-08:
Mambo calib do not work on simu and we don't have a mambo to try, can you confirm that calibration of pots occurs in main callib @Sekilsgs2  ? (code suggest it should)

Sekilsgs2, 2022-01-08:
> Mambo calib do not work on simu and we don't have a mambo to try, can you confirm that calibration of pots occurs in main callib @Sekilsgs2 ? (code suggest it should)

No, tbs have separate calibration menu ( **void menuCommonCalibPots(event_t event)** )for pots in 
**https://github.com/opentx/opentx/blob/2.3/radio/src/gui/128x64/radio_calibration_hall.cpp**
Need separate button for this.
see video with the right behavior - 
https://mega.nz/file/ELIXGQbZ#9zVXHGEdpKjlsYcxCXt_Aha93lzSl_m58xOq3yAq_fY

Need to add a **POTS_CALIBRATION** menu to
**void menuRadioHardware(event_t event)**
But with latest commits need litle rework current defines for **POTS_ROWS**

3djc, 2022-01-08:
I disagree that this is the right behaviors, it should be trigered after sticks are calibrated, if not at same time. The code suggest it is done using page key : https://github.com/opentx/opentx/blob/2.3/radio/src/gui/128x64/radio_calibration_hall.cpp#L76

Sekilsgs2, 2022-01-08:
> I disagree that this is the right behaviors, it should be trigered after sticks are calibrated, if not at same time. The code suggest it is done using page key : https://github.com/opentx/opentx/blob/2.3/radio/src/gui/128x64/radio_calibration_hall.cpp#L76

:-D
Yes page key list next pots calibration menu! I'm test and this working))
There is no information about this anywhere...
Thanks!

3djc, 2022-01-08:
Which means we need to remove that button, or make it working so leaving this issue opened

Sekilsgs2, 2022-01-08:
> Which means we need to remove that button, or make it working so leaving this issue opened

Yes i'm want say this too,but thought it might for others :)

3djc, 2022-01-08:
Could you try this branch ? It should have the button working (in addition of PAGE) : https://github.com/opentx/opentx/compare/3djc/tbs-pots-calib

Sekilsgs2, 2022-01-08:
> Could you try this branch ? It should have the button working (in addition of PAGE) : https://github.com/opentx/opentx/compare/3djc/tbs-pots-calib

Yes this works correctly!
Thanks.

PR #8817 (2022-01-13): Enables calib button for pots for tbs radio
To cope with 'unstandard TBS calib process

This fixes #8815
closed 2022-01-08 by Sekilsgs2