opentx/opentx #8830: Jumper T-Pro and companion 2.3.14, cannot read models and settings from radio, error: 'Can't extract RADIO/radio.bin'

karmensarodjo, 2022-01-16:
Dear All,

I'm new to OpenTX. Just bought a Jumper T-PRO and did a few things, ran into this problem right away. I've been trying to find a solution for almost two days and haven't found it yet, ...did I miss something, should do or download a more appropriate version? Thanks in advance!

Background,
- Download and prepare the sdcard file sdcard-128x64-2.3V0038.zip on the micro SD Card and insert it into the radio. Now radio can talk.
- Download OpenTX Companion, install, and added profile settings as Jumper T-Lite, the rest are default
- No problem with PC and radio connection. The SD Card radio is read by Explorer complete with its folders and there is another drive with some bin files.
- Everything is still original 
- ...haven't flashed anything to radio, just insert SD card, create and name a model, set it as FUTABA (no further setup).

Steps to reproduce the behavior:
1. Power up the radio via bootloader
2. Connect USB C cable
3. Run Companion
4. Goto Read/Write>Read Models and Settings From Radio

Expected behavior:
The process should be seamless to read and copy models and radio settings

Screenshots:
Like other 'Can't extract RADIO/radio.bin' screenshots

General information:
 - OpenTX version... (how can i check firmware version of Jumper T-PRO?)
 - Hardware brand new Jumper T-PRO, brand new SD Card microSDHC 16GB
 - OpenTX Companion 2.3.14 on Windows 7

unites, 2022-01-16:
Just pointing out this same thing is seen with EdgeTX.  I may open a similar issue over there.  I am guessing something needs to be added for the Jumper T Pro to both "companions".

FYI OP, if you hold down your menu button you are thrown into an Admin menu and the last page will give you info about your firmware.

3djc, 2022-01-16:
I'll fix that tomorrow

unites, 2022-01-16:
> I'll fix that tomorrow
@3djc 
I can create a separate issue for this but thought I would ask.  Do you or anyone here happen to have the Jumper T Pro firmware?  In the process of messing with this I accidently flashed the T-lite and am not able to find a version of the T-Pro firmware bin.

3djc, 2022-01-16:
You should be able to get a nightly

3djc, 2022-01-16:
Tested ok with 2.3.15N624, please ensure your current profile is a T-Pro

closed 2022-01-16 by 3djc

unites, 2022-01-16:
> Tested ok with 2.3.15N624, please ensure your current profile is a T-Pro

@3djc  Thanks, however... I am getting an "Invalid board" error when attempting to download any T-Pro firmware using the Companion app on nightly 2.3.15N624 for mac, windows and debian.  Is there a URL/URI I could access to just pull the bin file for the T pro?

3djc, 2022-01-16:
Have you set your channel to nighties ?

unites, 2022-01-16:
Ah, there is a channel drop down in the Companion app. I see it now on tab two of settings.  Apologies this is day 2 for me in this hobby, still learning my way around.  All set, new T-PRO firmware is now in place!  Thanks again!

karmensarodjo, 2022-01-16:
Thank you, all! …will be tested after office

monkfood, 2022-04-21:
Hello, I am facing the same issue. 
My aproach:

- Downloaded and installed nightly build 665
- Created a model for T-PRO, downloaded the firmware (trom nightly channel)
- Connected T-PRO RC to Laptop USB: It says that USb cable is connected on he RC, no reaction on the mac or a windows PC
-  Push firmware or model.

Error: No DFU Compatible device found

Is there anything i can do?