opentx/opentx #8839: Companion: t-pro switches missing

arkarkark, 2022-01-22:
**Describe the bug**

I'm unable to select SW1 or SA for special functions or anything really
also in the simulator the switches are missing. To be fair the factory firmware shipped with the T-pro doesn't seem to support the switches properly either.

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. edit a model on a t-pro radio
2. go to special functions
3. try and play a track when switch 1 is either on, off or in the middle position 
4. There is no way to select switch 1 or switch a whatever you want to call it

also in the simulator the switches and momentary buttons are missing.

[image]

or if you copy over a model from a t-lite that has something set to play a track for each of the three positions for switch A then it doesn't import correctly and the switch selectors are blank

[image]

**Expected behavior**

I should be able to make a track play for each of the three positions on the buttons and for each momentary trigger button and also for the 6 buttons above the display.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version N631
 - Hardware T-Pro
 - OpenTX Companion OSX

3djc, 2022-01-22:
Import and other improvement are part of N633

karmensarodjo, 2022-01-22:
hi 3djc,

i'm using 2.3.15N632 Jan.22 2022 right now, and the simulator switches are still missing, and the 6 little buttons are still disable...

3djc, 2022-01-22:
Yes, that's why it is still opened

arkarkark, 2022-01-23:
confirmed this is fixed in N633 Thanks so much!

3djc, 2022-01-23:
What is fixed and what isn't, because the 6switch on simu arent working, right ?

3djc, 2022-01-23:
I think those are still opened

[image]

[image]

arkarkark, 2022-01-23:
SA -> SD switch options are there in Companion.

Indeed, the 6 momentary switches are still not there. good call leaving this open. Sorry.