opentx/opentx #8841: T-pro switch SA stays up when switching to middle "-" position

arkarkark, 2022-01-23:
**Describe the bug**

I have special functions which play a different track depending on the three states of SA
on my T-Pro

when I switch from "UP" to "-" it doesn't play the middle sound. when I look at the special functions screen on my radio when I am in the "-" position SA "UP" is also lit. and it never plays the "-" sound.

**To Reproduce**
Steps to reproduce the behavior:
1. go to special functions
2. make a line for each of SA Up, -, Down and have each play a different track
3. now move switch SA and notice it never plays the middle sound
4. view the special functions screen and see that SA Up stays lit when you move SA to "-"

**Expected behavior**

SA"-" functions should be called when you switch to SA"-"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version N633 <- I flashed this to my T-Pro
 - Hardware Jumper T-Pro
 - OpenTX Companion OSX

pfeerick, 2022-01-23:
Just as another data point, not necessarily invalidating your issue, but with a self-build of the current 2.3 branch I have both up, mid and down positions responding correctly. What does the RADIO/SYS -> Hardware page say - does it indicate that SA is configured as a 3POS switch? And if you scroll down further and go into "Keys", does it show SA going through all three states?

3djc, 2022-01-23:
Can seem to reproduce this issue easily. Coud you go to hardware screen, and at the bottom, do an, EEprom backup. This will create a file in \EEPROM directory on your sd card. Please post this file here

arkarkark, 2022-01-23:
@pfeerick hit the nail on the head. I went to hardware and indeed my switches were all set to None. I set A and B  to 3 POS it it's working great now! Thanks! This also fixed the problem of the button state not showing up on the home screen.

I set C & D to Toggle.

Apologies for the false report.

closed 2022-01-23 by arkarkark

3djc, 2022-01-23:
If you have been using earlier version of companion, and do not have too much model, I suggest you wipe your eeprom and restart fresh. There was import/export issues previously