opentx/opentx #8842: Companion: T-Pro Logical Switches & Special Functions, blank and wrong...

karmensarodjo, 2022-01-23:
Hi All,

To be honest, i'm not sure if this is a bug or I'm missing something...
I'm still exploring programming in OpenTX before using it for my rc heli. And the lines of logic I've written below, are what I expected and work fine on my Jumper T-Pro radio. But when I read it to Companion to do a backup, it doesn't read correctly like the screenshot below. 
Please check... thanks in advance!

Steps to reproduce the behavior:
1. ...read Models and Settings from Radio
1. edit the model
2. go to Logical Switches and Special Functions

Expected behavior:
The Companion should reflect of my radio settings

Screenshots:
Logical Switches:
[image]

Special Functions:
[image]

General information:
OpenTX version 2.3.15N633
OpenTX companion-windows-2.3.15N633 on Windows 7

3djc, 2022-01-23:
Please save your eeprom with radio (hardware page, bottom, will be saved in EEPROM ssd directory) and paste it here

karmensarodjo, 2022-01-23:
@3djc,

plase find the attached file below...
[eeprom-2022-01-23-190634.zip](https://github.com/opentx/opentx/files/7920417/eeprom-2022-01-23-190634.zip)

thanks you!

3djc, 2022-01-25:
Fixed in 17af9c2328711fd34527f2c08e3697afd89fa0cf

closed 2022-01-25 by 3djc