opentx/opentx #8844: Companion is clobbering hardware switch selection

arkarkark, 2022-01-25:
**Describe the bug**

This might be a T-Pro only thing. When you write models and settings to radio, your hardware settings for the switches all get clobbered to None.

**To Reproduce**
Steps to reproduce the behavior:
1. read settings and models from T-Pro radio
2. write settings and models to radio
3. on radio go to the hardware page and look at the bottom to what your SA, SB... switches are set to. They are None on mine where before they were 3POS, 3POS, Toggle and Toggle

I did an restore to factory settings, set up my switches on my radio and then did steps 1-3 and observed the problem.

**Expected behavior**

reading settings and then writing those same settings shouldn't lose hardware settings.
This is bad because it also breaks any special functions or logical switches that also use SA, SB etc...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version N633
 - Hardware T-Pro
 - OpenTX Companion Mac N633

pfeerick, 2022-01-25:
As an addition to this - they are being read and displayed correctly by Companion, iit seems it's only when you go to write that it all goes horribly wrong.

3djc, 2022-01-25:
Yes companion should not be used right now with T-Pro, lots of issues

3djc, 2022-01-25:
Fixed in 17af9c2328711fd34527f2c08e3697afd89fa0cf

closed 2022-01-25 by 3djc