opentx/opentx #8846: QX7 ACCST intermittent binding in D8 mode after 2.3.14 update

fqahmad, 2022-01-27:
The Radio will loose binding after a long switch off. In radio setup if reselect D8 mode then it binds after a few seconds and remains bounded. However if switched off and on after 5 minutes, the binding is lost. Have to redo the above select procedures to bind again. I down graded to 2.3.11 from a backup and problem is gone.

paulo-sousa, 2022-05-26:
I had commented on issue #8899, which might be the same as this one, although that issue refers to an external XJT module, whereas Q X7 has an internal XJT module.

I upgraded from 2.3.10 to 2.3.15 and observed the same thing, but the binding was lost independently of the duration of the power off.

This is what I can observe:
 - I can bind correctly in D8 mode.
 - Once bound, the connection to the receiver works correctly until a power-off.
 - When powered back on, the radio does not connect to the receiver.
 - The issue is intermittent, but the failure is observed on the large majority of the tests (i.e., it only connects correctly some times).

alm-nl, 2022-05-31:
The issue is related to ACCST v1 firmware, with ACCST v2.1 the issue does not occur as can be read in: https://www.rcgroups.com/forums/showthread.php?3961323-QX7-wont-bind-to-D8-receivers-after-OpenTX-upgrade/page2

It could be a choice to make the Synced Pulses optional so users who do not want to use ACCST v2.1 can also use D8 protocol.

paulo-sousa, 2022-05-31:
Yes, for reference, I tested with a **Taranis Q X7** and firmware **XJT-Ver170317**.

I tested commits incrementally and found that it stopped working after merge request #8421.

The issue might not be related to the Synced Pulses implementation directly, but rather the reimplementation of the internal module serial driver that was done to support Synced Pulses. I tested that it worked up until [this fix](https://github.com/opentx/opentx/pull/8421/commits/e00224fea192e133898d887c6c1cf7f325c935b1) specifically.

Users can override the problem with a XJT module firmware upgrade, but the underlying problem is that users that were reluctant to upgrade to ACCST 2.x.x to avoid breaking compatibility with their existing D16 receivers, or even users that don't know about this incompatibility will simply see their system stop working when they upgrade OpenTX.