opentx/opentx #8854: Power on Delay Farb Radio

Devil-67, 2022-02-02:
Please add the option Power on Delay for colour radios as for black and white.
[image]
[image]

3djc, 2022-02-02:
Power on delay is dictated by the hardware loading time which are much longer on colorlcd radio

closed 2022-02-02 by 3djc

Devil-67, 2022-02-02:
I'm only concerned with the time it takes to hold the power button, not the time it takes to start.

3djc, 2022-02-02:
Not sure I understand, it takes a 1sec push, but thats hardware, you want it longer ?

Devil-67, 2022-02-02:
No I want it shorter, so just tap it.
The best way to adjust it is the way it is with X9D 2019.
I use 0 sec on the X9D

3djc, 2022-02-02:
Then that won't happen, since the only delay there is is hardware

Devil-67, 2022-02-02:
hmm... 
ok, then you can close it

Thanks