opentx/opentx #8858: Elrsv2.lua script - script panic not enough memory

lion13ch, 2022-02-03:
I use OpenTX 2.3.14(X9D_OneBit_Status_2_3_14.bin) at Taranis X9D with HM ES24TX Pro module. Elrsv2.lua script run once normal but the other times it get memory error (script panic not enough memory) until next radio off/on. Is there any way to fix it?
Without OneBit option Taranis X9D runs Elrsv2.lua script very slow (slow inverter problem).

3djc, 2022-02-03:
Don't think onebit is something we do support...

lion13ch, 2022-02-03:
It is the custom version based on opentx 2.3.14 from www.expresslrs.org (https://heatermeter.com/dl/elrs/).
Is there any way to get native OneBit support for OpenTX? Or maybe just go ti EgdeTX?

ChillingChad, 2023-01-05:
i have found if you get edgeTx 2.7.1 the lua script work perfect but at 2.8 it does not

closed 2023-01-05 by 3djc

cmanley, 2023-02-10:
> i have found if you get edgeTx 2.7.1 the lua script work perfect but at 2.8 it does not

Same here: the lua script fails with the panic message on an X9D using EdgeTX 2.8.0

jcink, 2023-02-17:
I have the same problem on a pre-2019 XD9+ on 2.8, ELRSV3 lua script fails with the panic message about memory. I also cannot load my iNav telemetry script either, INAV LUA 2.2.0. I downgraded to 2.7.2 and the scripts load. So it would seem 2.8 is a little memory hungry? Any thoughts on this? I noticed that with edgetx flasher, the firmware does not seem to be compiled with "noheli"... if it was, could that help to free some space to resolve this issue?

cmanley, 2023-02-17:
I guess the solution from the ELRS devs point of view would be to chop the LUA script into multiple library scripts as is done by some other larger LUA widgets. What that'll do (if my guess is correct) is cause the LUA compiler to have to compile smaller chunks of code at a time, thereby consuming less memory. After all it's the compile phase that's the problem and not the run phase, as I ran the LUA script on a different radio (the Jumper T-Lite) and then copied the compiled *.luac file over to the X9D and it worked which surprised me a bit because I expected the compiled file to be hardware dependent.

3djc, 2023-02-17:
> I guess the solution from the ELRS devs point of view would be to chop the LUA script into multiple library scripts as is done by some other larger LUA widgets. What that'll do (if my guess is [...]

It is hardware dependent. Extremely generic lua might work that way, but if radio dependant stuff is used, it will malfunction. But if you have issues with EdgeTx, it really should be reported on EdgeTx GitHub ....

cmanley, 2023-02-17:
Link to issue in ELRS github:
https://github.com/ExpressLRS/ExpressLRS/issues/2052